wpilibsuite/allwpilib #5885: CMake errors when building ntcore beta 3

binex-dsk, 2023-11-06:
**Describe the bug**
When building only ntcore `WITH_NTCORE=ON` and everything else `OFF`, everything compiles as expected. However, attempting to use ntcore in a CMAKE project with `find_package(... wpilib REQUIRED)` results in the following error during CMake configuration:

```
CMake Error at /usr/share/cmake/Modules/CMakeFindDependencyMacro.cmake:76 (find_package):
  By not providing "FindwpilibNewCommands.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "wpilibNewCommands", but CMake did not find one.

  Could not find a package configuration file provided by "wpilibNewCommands"
  with any of the following names:

    wpilibNewCommandsConfig.cmake
    wpilibnewcommands-config.cmake

  Add the installation prefix of "wpilibNewCommands" to CMAKE_PREFIX_PATH or
  set "wpilibNewCommands_DIR" to a directory containing one of the above
  files.  If "wpilibNewCommands" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /usr/share/wpilib/wpilib-config.cmake:14 (find_dependency)
  CMakeLists.txt:14 (find_package)
  ```
  
  Removing the offending line in the system cmake file (why is it there in the first place?) results in a new error regarding protobuf:
  
  CMake Error at /usr/share/wpiutil/wpiutil.cmake:60 (set_target_properties):
  The link interface of target "wpiutil" contains:

    protobuf::libprotobuf

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /usr/share/wpiutil/wpiutil-config.cmake:12 (include)
  /usr/share/cmake/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  /usr/share/wpilib/wpilib-config.cmake:7 (find_dependency)
  CMakeLists.txt:14 (find_package)
  ```
  
  protobuf is installed.

All of this only occurs with 2024 beta 3. Beta 2 works perfectly.

binex-dsk, 2023-11-06:
Looks like using `ntcore` instead of `wpilib` for the CMake package fixes the first error, but the second remains.

PeterJohnson, 2023-11-06:
I think the only cmake changes between beta-2 and beta-3 were #5797 and #5768.  Does reverting either of these fix the build for you?  What about setting `WITH_JAVA_SOURCE=OFF`?

calcmogul, 2023-11-06:
Do we know if this ever worked?

PeterJohnson, 2023-11-06:
> All of this only occurs with 2024 beta 3. Beta 2 works perfectly.

binex-dsk, 2023-11-06:
Nope, reverting those and/or setting `WITH_JAVA_SOURCE=OFF` did not work.

As @PeterJohnson said, yes, this worked fine on Beta 2

calcmogul, 2023-11-06:
The only CMake changes to wpiutil, wpinet, and ntcore in beta 3 were the Java source jar thing. `WITH_JAVA_SOURCE=OFF` effectively removes the code for that. https://github.com/wpilibsuite/allwpilib/pull/5797 didn't impact any of those three libraries.

binex-dsk, 2023-11-06:
Looks like something changed on my system, as beta 2 doesn't work anymore... However, the fact that it occurred on both systems after upgrading ntcore (and no other packages were upgraded)  is awfully suspicious. I'll do a bit more testing.

calcmogul, 2023-11-06:
https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/wpiutil-config.cmake.in

I'm not sure why the config file does things that way. The official docs take a different approach: https://cmake.org/cmake/help/book/mastering-cmake/cmake/Help/guide/tutorial/index.html#adding-export-configuration-step-11

binex-dsk, 2023-11-06:
Adding `find_dependency(protobuf)` to `wpiutil-config.cmake.in` fixes this as well, I'll make a PR adding that.

@calcmogul which part of the cmake file are you talking about? I can probably add that to my PR.

calcmogul, 2023-11-06:
The whole file in general. Cleaning that up belongs in a separate PR.

closed 2023-11-06 by PeterJohnson