wpilibsuite/allwpilib #5901: JSON parsing errors when running C++ on RoboRIO

mjansen4857, 2023-11-09:
**Describe the bug**
I've received a couple reports now of people having trouble parsing PathPlanner json files on the robo rio when using C++ since the recent changes to the C++ json library. I am unable to reproduce this issue in simulation, so it seems to be specific to the rio, but I do not currently have access to one I can use for further testing. 

The errors being generated are syntax errors, yet there are no syntax errors with the JSON files. It seems to be failing on the closing `}` at the end of the file. These files can be parsed without issue with other json libraries, and I am parsing the files identically to how other json files are parsed throughout WPILib, so I don't think this is an error on my end.

For example, here is how auto files get parsed in PathPlannerLib:
```C++
std::error_code error_code;
std::unique_ptr < wpi::MemoryBuffer > fileBuffer = wpi::MemoryBuffer::GetFile(filePath, error_code);

if (fileBuffer == nullptr || error_code) {
	throw std::runtime_error("Cannot open file: " + filePath);
}

wpi::json json = wpi::json::parse(fileBuffer->begin(), fileBuffer->end());
```

This does not use the new `GetCharBuffer` method from #5877, which is unreleased. I'm curious if this would have any affect on this issue?

Perhaps it is also possible that #5875 would have an effect here as well?

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy this file to the robo rio in a C++ robot code project: [test.auto](https://drive.google.com/file/d/1t6eUf48tzGIK6N5Dfc7AmnxUk2kZVHKd/view?usp=sharing)
2. Attempt to parse the file with the wpi json library

**Expected behavior**
JSON library should parse valid json files without syntax errors

**Screenshots**
Here are error output screenshots provided by users:
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 11]
 - Project Information:
 ```
WPILib Information:
Project Version: 2024.1.1-beta-3
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    PathplannerLib (2024.0.0-beta-5)
    WPILib-New-Commands (1.0.0)

```

PeterJohnson, 2023-11-09:
This is almost certainly due to the issue fixed by 5875, because prior to that, there would be trailing zeros in the buffer.

spacey-sooty, 2023-11-30:
We've been experiencing an issue that might be related to this. Our simulation (running on WSL) is crashing because of an invalid end of input from a pathweaver json file.

PeterJohnson, 2023-11-30:
I'm going to close this, as it should be fixed by #5875.