wpilibsuite/allwpilib #5908: ArmSimulation test fails on GCC 13

calcmogul, 2023-11-10:
```
[----------] 3 tests from ArmSimulationTests/ArmSimulationTest
[ RUN      ] ArmSimulationTests/ArmSimulationTest.Teleop/75deg
Not loading CameraServerShared
NT: Listening on NT3 port 1735, NT4 port 5810

********** Robot program startup complete **********
/home/tav/frc/wpilib/allwpilib/wpilibcExamples/src/test/cpp/examples/ArmSimulation/cpp/ArmSimulationTest.cpp:53: Failure
Expected equality of these values:
  kDefaultArmSetpoint.value()
    Which is: 75
  frc::Preferences::GetDouble(kArmPositionKey, (__builtin_nanf ("")))
    Which is: 50

Default RobotPeriodic() method... Override me!
Default DisabledPeriodic() method... Override me!
[  FAILED  ] ArmSimulationTests/ArmSimulationTest.Teleop/75deg, where GetParam() = 8-byte object <00-00 00-00 00-C0 52-40> (107 ms)
```
The arm simulation tests haven't changed, so maybe it's a race condition in NT the test was relying on turning out a certain way?

narmstro2020, 2023-11-17:
I've noticed this too.  Though my numbers are 75 and 25.  

Just a note for anyone else running into this.  If running a local build for testing on a completely unrelated piece of code.  I commented out the line.  ran the build and before committing uncommented the line back in.

Starlight220, 2023-12-20:
I suspect this is an issue with resetting preferences not being properly reset between test cases.

narmstro2020, 2024-10-11:
So the Preferences key check (line 72 of java file) still seems to have a problem with it.  I ran the test on my end for PR #7170 multiple times with no crashing.  Today I got one at line 72.