wpilibsuite/allwpilib #5915: Make ProfiledPIDController support arbitrary profile types

calcmogul, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
WPILib now has trapezoid and exponential profiles, and it would be nice to have a more generic interface for them in ProfiledPIDController.

**Describe the solution you'd like**

1. Make an interface in Java and a concept in C++ for motion profiles.
2. Add suppliers to the ProfiledPIDController interface.

auscompgeek, 2023-11-11:
> Add suppliers to the ProfiledPIDController interface.

This would be simpler for Python to wrap.

Gold856, 2023-11-12:
If you use suppliers, how do you change the constraints? `TrapezoidProfile` and `ExponentialProfile` both use different constraints.