wpilibsuite/allwpilib #5920: [gradle] Add generated protobuf headers to C++ test include path

pjreiniger, 2023-11-12:
**Is your feature request related to a problem? Please describe.**
Currently generated protobuf headers are not put into the include path that is used to build the tests. I tried a couple things to fix add the include path but didn't make any progress and thought Thad might now how to fix it real quick.

To reproduce, add a test file like this into `wpimath/src/test/native/cpp/geometry`

```
#include <gtest/gtest.h>
#include "geometry.pb.h" // Gets a file not found error

TEST(IncludeTest, Test) {
    wpi::proto::ProtobufPose2d proto;
   proto.mutable_translation()->set_x_meters(5);
   std::cout << proto.DebugString() << std::endl;
}
```

pjreiniger, 2023-11-16:
I swore I had tried this and hoped you did it slightly differently than me, but this is what I was getting

https://github.com/wpilibsuite/allwpilib/actions/runs/6887451508/job/18734682127?pr=5918#step:6:3700

This shoudl be reopened

KangarooKoala, 2023-11-16:
The specific error message is this (lines 3699-3719 of the posted CI link):
```
Some problems were found with the configuration of task ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' (type 'CppCompile').
  - Gradle detected a problem with the following location: '/work/wpimath/build/generated/source/proto/main/cpp/geometry2d.pb.h'.
    
    Reason: Task ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' uses this output of task ':wpimath:generateProto' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed.
    
    Possible solutions:
      1. Declare task ':wpimath:generateProto' as an input of ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp'.
      2. Declare an explicit dependency on ':wpimath:generateProto' from ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' using Task#dependsOn.
      3. Declare an explicit dependency on ':wpimath:generateProto' from ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' using Task#mustRunAfter.
    
    For more information, please refer to https://docs.gradle.org/8.4/userguide/validation_problems.html#implicit_dependency in the Gradle documentation.
  - Gradle detected a problem with the following location: '/work/wpimath/build/generated/source/proto/main/cpp/geometry3d.pb.h'.
    
    Reason: Task ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' uses this output of task ':wpimath:generateProto' without declaring an explicit or implicit dependency. This can lead to incorrect results being produced, depending on what order the tasks are executed.
    
    Possible solutions:
      1. Declare task ':wpimath:generateProto' as an input of ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp'.
      2. Declare an explicit dependency on ':wpimath:generateProto' from ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' using Task#dependsOn.
      3. Declare an explicit dependency on ':wpimath:generateProto' from ':wpimath:compileWpimathTestReleaseGoogleTestExeWpimathTestCpp' using Task#mustRunAfter.
    
    For more information, please refer to https://docs.gradle.org/8.4/userguide/validation_problems.html#implicit_dependency in the Gradle documentation.
```
So I think the error is about Gradle configuration, not whether the protobuf headers are in the C++ include path.

PeterJohnson, 2023-11-16:
Ah, we need to add a dependency linkage... should be pretty straightforward to do that in shared/jni/setupBuild.gradle, but I'm not sure of the exact Gradle incantation to do it.

PeterJohnson, 2023-11-16:
#5933 should fix that issue.

PR #5926 (2023-11-16): [build] Add generated protobuf headers to C++ test include path
Fixes #5920.
PR #5933 (2023-11-17): [build] Add generateProto dependency to test and dev
Fixes #5920.
closed 2023-11-17 by PeterJohnson