wpilibsuite/allwpilib #5921: C++ command-based examples use both `Command*` and `CommandPtr` for `GetAutonomousCommand`

rzblue, 2023-11-13:
The examples show `GetAutonomousCommand` returning both `CommandPtr` and `Command*` in various places.

In some examples, `GetAutonomousCommand` constructs and returns a `Command*` constructed with the `new` operator, leaking memory.
In others, it returns pointers to commands constructed inside `RobotContainer` (which obviously can't be accomplished with `CommandPtr` due to ownership constraints- functions that return `CommandPtr` must construct a new command each time)
)

Additionally, `SendableChooser` *must* be used with `Command*`.

It seems like there's not one way that will work for all cases.

I'm not sure what the best solution is here. Perhaps we should keep the examples as-is, and better document the differences between `CommandPtr` and `Command*`?

Starlight220, 2023-11-13:
> In some examples, GetAutonomousCommand constructs and returns a Command* constructed with the new operator, leaking memory.

This is a leftover from before CommandPtr was a thing (and possibly even from old commands).

> it returns pointers to commands constructed inside RobotContainer (which obviously can't be accomplished with CommandPtr due to ownership constraints- functions that return CommandPtr must construct a new command each time)

`CommandPtr` or `Command*`? Returning `Command*` from a `CommandPtr` owned elsewhere with the correct lifetime (such as RobotContainer) is ok. As long as ppl don't start composing it. 

> Additionally, SendableChooser must be used with Command*.

That's been brought up before; there are multiple options:
- Choose between Command*, own the commands externally in RobotContainer
- Choose between `function<CommandPtr()>`.
- Choose between strings/ints/enums/etc, feed them into SelectCommand. (Maybe add a class for doing this?)

> better document the differences between CommandPtr and Command*?

I think we have some docs in the C++ Commands page, but improving it and/or putting it somewhere more obvious is never a bad idea.
Having concepts might improve this, might make it more confusing due to a bunch of type names.

Starlight220, 2023-12-02:
To clarify, the action items I see from this issue are:
- Eliminate the memory leaks from `new`ing auto commands (this can be done by owning the command in `RobotContainer`, or constructing and passing a `CommandPtr`).
- Document differences between `CommandPtr` (owned) and `Command*` (non-owned).
- Better showcase usage of how to use `CommandPtr` in conjunction with `SendableChooser` (preferably, have examples for all three ways).

Using concepts for command types should also be done, but I see that as deriving more from #5789 than from this issue.

Starlight220, 2023-12-03:
The C++ Commands Technical Discussion page on frc-docs, though it might need a refactor/split/rework.

Starlight220, 2023-12-14:
I extracted the remaining documentation part to wpilibsuite/frc-docs#2438; closing.

closed 2023-12-14 by Starlight220