wpilibsuite/allwpilib #5929: [wpimath] MecanumDriveWheelVoltages is only in Java

KangarooKoala, 2023-11-16:
**Describe the bug**
Java has a `MecanumDriveWheelVoltages` class and C++ does not. In wpilib, the only place `MecanumDriveWheelVoltages` is used is in `MecanumControllerCommand`, where it is used as a parameter type to the callback to set the motor voltages (`Consumer<MecanumDriveWheelVoltages>`). The C++ version instead uses 4 separate parameters for the callback (`std::function<void(units::volt_t, units::volt_t, units::volt_t, units::volt_t)>`).

**Expected behavior**
C++ and Java either both have or both don't have `MecanumDriveWheelVoltages`, and their `MecanumControllerCommand` APIs are similar.

PeterJohnson, 2023-11-16:
If it's only in that one place, we could replace the Supplier with a `@FunctionalInterface` that takes the 4 parameters, just like C++.

KangarooKoala, 2023-12-05:
Although it would be fairly easy to migrate, I don't think it would be good to introduce a breaking change like this one so close to kickoff (1 month left, with winter break for most schools in between). Maybe a 2025 goal? (Or maybe there is still enough time for this since it's so simple?)

PR #6760 (2024-12-09): [wpilib] Replace MecanumDriveMotorVoltages with a functional interface
Resolves #5929
closed 2024-12-09 by calcmogul