wpilibsuite/allwpilib #5937: [wpimath] Protobuf symbols aren't exported on windows

pjreiniger, 2023-11-17:
wpimath does explicit symbol exporting unlike the other libraries. Since the generated C++ do not contain `WPILIB_DLLEXPORT`, building against a dll will get linking errors. I created a new [branch](https://github.com/wpilibsuite/allwpilib/compare/main...pjreiniger:allwpilib:protobuf_old_serde_tests_only?expand=1) with _just_ the test changes which fails locally and should fail [here](https://github.com/pjreiniger/allwpilib/actions/runs/6907511912)

I was building all these unit tests on wsl so I didn't notice the problem until Peter fixed the include path thing.

mcm001, 2023-11-19:
This also happens with a vendordep which depends on native protobufs from allwpilib. We could potentially hack around this by copy pasting the messages into our source tree?

srimanachanta, 2023-11-30:
Can this be closed? PV has seen this resolved with recent PRs to wpilib.

PeterJohnson, 2023-11-30:
Fixed by #5957.

closed 2023-11-30 by PeterJohnson