wpilibsuite/allwpilib #5940: C++ Sendable Chooser template arguement requirements

NathanCorrea24, 2023-11-20:
**Describe the question you have.**
Why does the sendable chooser require the template arguement to be copy constructible? 

**Describe the reason for your confusion.**
In the implementation of sendable chooser it only uses the move constructor so it seems odd that it is required to be copy constructible. 

**Is your question related to a problem? Please describe.**
I'd like to be able to use frc2::CommandPtr in the sendable chooser which is not copy constructible but it is move constructible. Would it be possible to require the template arguement to be std::is_move_constructible or std::is_nothrow_move_constructible instead of requiring the copy constructor to exist?

Starlight220, 2023-11-20:
> In the implementation of sendable chooser it only uses the move constructor

Can you back this? Last time I checked this wasn't the case; it also doesn't logically make sense with the idea of SendableChooser.

Starlight220, 2023-11-20:
The options are moved into the chooser, but they're copied out (see `GetSelected`) -- otherwise you'd have invalid objects at the second call.

ncorrea210, 2023-11-21:
I guess I'm wondering then why the Sendable Chooser does not have command pointers in mind then? If the goal is to make C++ programming safer it seems like encouraging the use of CommandPtr over Command* would be good however CommandPtr is not and cannot be compatible with Sendable Chooser in its current states as a unique_ptr wrapper. I think some of this is mentioned in the issue you referenced before, #5921.

PeterJohnson, 2023-11-21:
We can change GetSelected to return a const(?) ref instead of a copy, but I’m not sure that’s really all that useful of a change, as the caller would probably need to copy it anyway?

Starlight220, 2023-11-21:
A const ref of CommandPtr won't help.

ncorrea210, 2023-11-21:
If the CommandPtr wrapped a shared pointer than it could be made to be copy constructable. Not sure if it defeats the purpose of CommandPtr but that could make it work for SendableChooser.

Starlight220, 2023-11-22:
That would defeat the point of CommandPtr.

I've already illustrated three methods for doing this. You are right in that docs/examples should use these solutions (and the possible addition of a ChooserCommand), but that's already in the realm of #5921 and not in the scope of this issue.

I don't see any action items left here.

closed 2023-11-22 by Starlight220