wpilibsuite/allwpilib #5963: Extension of LoggedRobot and IterativeRobotBase in most recent 2024 version of WPILib Unrunnable

ajmoritz2, 2023-11-25:
**Describe the bug**
The bug is as simple as it sounds (not sure how simple to solve though). Whenever the program is ran, an error presents itself before even the Robot class is initialized. The error is as follows:
```
Exception in thread "main" java.lang.NoSuchFieldError: Unknown
        at edu.wpi.first.wpilibj.DriverStation.<clinit>(DriverStation.java:606)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:419)
        at frc.robot.Main.main(Main.java:7)
```

Line 7 of our Main class is as follows: 
`RobotBase.startRobot(Robot::new);`
This could possibly be an issue with a vendor library, but because these are WPILib classes, I'm not sure which one it would be.
This is all of the information I have

**To Reproduce**
Steps to reproduce the behavior:
1. Use most recent WPILib 2024
2. Install latestest vendordeps libraries
3. Run the robot code

**Expected behavior**
I expect the Robot class to start executing its code.

**Desktop Info**
- OS: Windows 10
- WPILib Information:
Project Version: 2023.4.3
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    AdvantageKit (3.0.0-beta-4)
    PathplannerLib (2024.0.0-beta-5.1)
    CTRE-Phoenix (v6) (24.0.0-beta-3)
    WPILib-New-Commands (1.0.0)

**Additional info**
So far I have only tested current code in simulation, but I was getting this same error on the robot with different code.

PeterJohnson, 2023-11-25:
Your project version is 2023.4.3.  You need to run the project importer in 2024 vscode to upgrade your project to 2024.

ajmoritz2, 2023-11-25:
Ah I was wondering about that one. That would make sense wouldn't it.

ajmoritz2, 2023-11-25:
That seemed to be the issue. Thank you for dealing with my silly mistake.

PeterJohnson, 2023-11-25:
Happy to help!

closed 2023-11-25 by PeterJohnson