wpilibsuite/allwpilib #5965: [ntcore] nt::Topic::SetProperty does not support infinity value in C++, but works in Java

virtuald, 2023-11-26:
Reported by RobotPy user in https://github.com/robotpy/mostrobotpy/issues/20.

To verify this is a C++ bug, I wrote a C++ function that directly did `Topic.SetProperty("foo", -INFINITY)` and on the remote side it displayed the property value as `null` as reported by the user.

PeterJohnson, 2023-11-26:
I’m not sure how it directly “works” in Java, since setProperty() in Java only accepts a string (which must be json encoded by the user), not a double.

PeterJohnson, 2023-11-26:
Doing a bit more digging, JSON does not support infinity or NaN values. They’re passed through by Java because it’s just dumping in the raw string into the properties value, but C++ is actually processing them as “real” (checked) json and thus rejects infinity values.

PeterJohnson, 2023-11-26:
It’s in accordance with the nlohmann json docs: https://json.nlohmann.me/features/types/number_handling/#nan-handling

auscompgeek, 2023-11-26:
> FWIW some JSON implementations accept infinity as a valid value (for example, Python will happily encode/decode infinity/NaN).

Yes, this is explicitly documented as violating the JSON spec. This is not interoperable.

> If allow_nan is false (default: True), then it will be a ValueError to serialize out of range float values (nan, inf, -inf) in strict compliance of the JSON specification. If allow_nan is true, their JavaScript equivalents (NaN, Infinity, -Infinity) will be used.

https://docs.python.org/3/library/json.html#json.dump

closed 2023-11-27 by PeterJohnson