wpilibsuite/allwpilib #5980: [cmake, ntcore] Failure to build on MinGW on Windows

binex-dsk, 2023-11-30:
**Describe the bug**
When attempting to build ntcore on MinGW on Windows with cmake, using the following commands:

```
cmake -B wpilib_build_win -S "allwpilib" \
          -DCMAKE_INSTALL_PREFIX='/usr' \
          -DUSE_SYSTEM_EIGEN=OFF \
          -DUSE_SYSTEM_FMTLIB=OFF \
          -DWITH_JAVA=OFF \
          -DWITH_CSCORE=OFF \
          -DWITH_NTCORE=ON \
          -DWITH_WPIMATH=OFF \
          -DWITH_WPILIB=OFF \
          -DWITH_TESTS=OFF \
          -DWITH_GUI=OFF \
          -DWITH_SIMULATION_MODULES=OFF \
          -DCMAKE_BUILD_TYPE=Release \
          -Wno-dev

cmake --build wpilib_build_win
```

Various errors occur: 
```
.../allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:108:3: error: 'EXPORT_TEMPLATE_TEST_MSVC_HACK_DEFAULT' was not declared in this scope; did you mean 'EXPORT_TEMPLATE_DECLARE_DEFAULT'?
  108 |   EXPORT_TEMPLATE_##which##_##style(export, )
      |   ^~~~~~~~~~~~~~~~

...

.../allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:108:45: error: expected primary-expression before ')' token
  108 |   EXPORT_TEMPLATE_##which##_##style(export, )

.../allwpilib/wpiutil/src/main/native/include/wpi/SymbolExports.h:189:33: error: expected primary-expression before '__attribute__'
  189 | EXPORT_TEMPLATE_TEST(MSVC_HACK, __declspec(dllexport));
      |                                 ^~~~~~~~~~
```
I have reasonable suspicion that this is caused by incorrect usage of the `_WIN32` and `_MSC_VER` macros, but I can't say for sure.

**Desktop (please complete the following information):**
 - OS: GH Actions windows-latest (Windows 11?)

**Additional context**
Note that this is done with Github Actions.

ThadHouse, 2023-11-30:
We don't support MinGW on windows, and have no plans to support it in the future.

calcmogul, 2023-11-30:
We only support MSVC compilers shipped with Visual Studio 2022.

closed 2023-11-30 by calcmogul

PeterJohnson, 2023-11-30:
We will likely accept PRs changing _WIN32 to _MSC_VER as appropriate, but I'm sure there are other issues, as we never build test with MinGW.

calcmogul, 2023-11-30:
The ifdefs at the top of SymbolExports.h seem correct (i.e., it has branches for Windows, then MSVC vs GCC). The test macros at the bottom may need conditionals.