wpilibsuite/allwpilib #5987: [wpiutil] MSVC: Failed to load entry point

binex-dsk, 2023-12-02:
**Describe the bug**
When compiling wpiutil with MSVC and launching an application linked to it, the following error window pops up.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Build wpiutil with CMake
2. Link the wpiutil dll to an application

 - Link to code: irrelevant?

**Expected behavior**
The application should run without issue but has the error above

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
wpiutil was built with CMake.

binex-dsk, 2023-12-02:
Here's the relevant CMake commands. https://github.com/binex-dsk/QFRCDashboard/blob/arg/.github/workflows/build.yml#L200

Forgot to mention this is on a recent commit (with my protobuf CMake PR)

Gold856, 2023-12-02:
Can you try building wpilib in debug mode? It says your app is built in debug mode, but you built wpilib in release mode. I used my cloned wpilib and built debug DLLs and it stopped complaining about the entrypoint, although I did hit another error from there.

binex-dsk, 2023-12-02:
Good catch! I completely forgot to change this option
`
 -DCMAKE_BUILD_TYPE=Release
 `

Gold856, 2023-12-02:
Remember that debug files have `d` for debug appended to the filename, so you'll also need to change the name in the workflow to match the filenames.

binex-dsk, 2023-12-02:
Yeah so far I've learned that MSVC is extremely stingy with debug/release files compared to gcc and clang. 

Enabling debug builds gives me this error during CI though:

```
D:\a\..\allwpilib\wpiutil\src\main\native\cpp\Logger.cpp: fatal error C1041: cannot open program database 'D:\a\...\win-nt-build\wpiutil\CMakeFiles\wpiutil.dir\vc140.pdb'; if multiple CL.EXE write to the same .PDB file, please use /FS
```

Don't know why it's only on debug builds? Enabling /FS does nothing

Gold856, 2023-12-02:
Maybe you need to change the target triplet?

binex-dsk, 2023-12-02:
I removed vcpkg entirely and that didn't help.

Gold856, 2023-12-02:
Alternatively, you could build everything in release mode, including the dashboard. That should mitigate the problem.

binex-dsk, 2023-12-02:
> You could also try doing a local build just to make sure that works.

I'm not able to do a local build, because Windows sucks and the only computer I can even use Windows 10 on has an installation that is so utterly garbage and broken that compiling any application on it would be a miracle on the same level as the resurrection of Christ. Let alone compiling two libraries and a dashboard, installing Qt, and getting MSVC to even function. This operating system is so garbage I genuinely don't understand how every single person at my school besides me uses this piece of trash operating system. 

Yet because of that, I'm forced to support this developer-hostile asinine piece of trash that can't even get its own installer to work properly, let alone even one aspect of their horrific operating system. If I don't support this garbage, I don't get a userbase. Period.

Anyways...
Switched to MSBuild and ntcore does finally build (with an added /EHsc flag to MSVC), but install failed with:
```
CMake Error at win-nt-build/wpiutil/cmake_install.cmake:89 (file):
  file INSTALL cannot find
  "D:/a/QFRCDashboard/QFRCDashboard/win-nt-build/bin/Release/wpiutil.dll":
  File exists.
```
...why is it looking for a release library... in a DEBUG build? I EXPLICITLY tell this john to build in DEBUG mode and it's out here looking for NONEXISTENT RELEASE LIBRARIES. I hate Windows so much.

Anyways, switching to release mode keeps the issue. I am so done rn :/

PeterJohnson, 2023-12-02:
I'm not sure how you're setting it to a DEBUG build, but I'm pretty sure CMAKE_BUILD_TYPE is ignored with msbuild.

MSVC does require keeping debug and release builds separate--they're not ABI compatible.

I also doubt that wpilib's cmake install has ever been tested on Windows.  It's definitely not a common combination for us.

binex-dsk, 2023-12-02:
> but I'm pretty sure CMAKE_BUILD_TYPE is ignored with msbuild.

Wonderful... How would I go about setting msbuild to debug mode with CMake? All I can find are passing arguments to msbuild directly which seems to be unsupported by CMake.

binex-dsk, 2023-12-02:
Nevermind, finally got this john to work. Thanks for the help @PeterJohnson @Gold856

closed 2023-12-02 by binex-dsk