wpilibsuite/allwpilib #6036: glass ignores json files

truher, 2023-12-12:
**Describe the bug**
Glass writes window state to json files but it erases that state when it tries to read it.

**To Reproduce**
Steps to reproduce the behavior:

1. open glass, fiddle with it
2. "save as global" --> the json files are correct
3. turn it off and on again
4. all the state is gone, the json files are reset

Please see  https://github.com/truher/glass_issues/blob/main/demo/README.md

**Expected behavior**
All the window state, plot state, etc, should be loaded instead of erased.

**Screenshots**
There are some screenshots in the repo:

https://github.com/truher/glass_issues/blob/main/demo/

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.3
 - Project Information: 
 - WPILib Information:
 - 
Project Version: 2024.1.1-beta-3
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.25.1
Java Debug Extension Version: 0.55.0
Java Dependencies Extension Version 0.23.3
Java Version: 17
Java Location: /home/joel/wpilib/2024/jdk
Vendor Libraries:
    WPILib-New-Commands (1.0.0)

**Additional context**

CD thread here:

https://www.chiefdelphi.com/t/sim-glass-gui-state/446006/2

truher, 2023-12-12:
playing with it some more, it's not consistent. sometimes it remembers some parts of the config, and sometimes it forgets everything.

it sometimes forgets the joystick-keyboard mapping, for example.

sometimes it remembers that there's a plot, but not its size or location, so it makes a tiny little box.

i haven't been able to reliably repro these cases; they seem to happen 10-20% of the time.

truher, 2023-12-12:
is there some sort of log where glass/simgui might complain about things like invalid json input, or something?  maybe it just erases things it doesn't understand, and the understanding depends on little details of the file that might be different in linux than in windows?

PeterJohnson, 2023-12-12:
We do report JSON parse errors to the imgui log, but I forget if we actually provide a way to view the imgui log in a window.  https://github.com/wpilibsuite/allwpilib/blob/main/glass/src/lib/native/cpp/Context.cpp#L169

truher, 2023-12-12:
oh, that sounds promising, but i can't find any way to view the log in the app, nor is it obvious where the file is written.  is there a place you specify the filename?

PeterJohnson, 2023-12-12:
If there's not a view for it, I bet it just goes to the bit bucket right now.

truher, 2023-12-12:
:(

truher, 2023-12-16:
I poked at this some more and confirmed that the json part is working, the ini file is fine, but something happens between the reading and the window.  I noticed this issue, https://github.com/ocornut/imgui/issues/6982 and I wonder if we're seeing the effect of it?  I tried building glass with imgui source but couldn't figure out how to make it work, is there a recipe for that?

PeterJohnson, 2023-12-16:
The easiest way to build with a different imgui version is to use cmake and modify imgui/CMakeLists.txt to use a different GIT_TAG.  Building a different version with Gradle is much more involved--you have to first modify thirdparty-imgui and publish it locally, then modify shared/imgui.gradle to point to it.

PeterJohnson, 2023-12-16:
I think I may revert the imgui upgrade if we can’t isolate the issue quickly.  There’s nothing really pressing we need in the new version.

truher, 2023-12-16:
oh!  were you able to reproduce it?  i assumed this was some sort of uncommon thing that is only happening to me.

anyway i'm working on it now, using cmake and a local copy of imgui.  i'm optimistic i'll find something today.

truher, 2023-12-16:
OK, submitted PR 6056.  The fix implies that it was always broken, which seems impossible.

Anyway, now I can have the library of task-specific configs that I want, which is glorious.  :-)

closed 2023-12-23 by PeterJohnson