wpilibsuite/allwpilib #6045: Make PIDController and ProfiledPIDController throw exceptions for negative feedback gains

calcmogul, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
Some users, and even WPILib's GearsBot example, use negative PID gains. The class assumes a positive output for positive error, so negative gains are unstable. Negating the output works around the issue, but it's not the correct fix.

**Describe the solution you'd like**
Constructors should throw an exception if negative feedback gains are provided. If negative gains are provided to SetPID(), etc., we should probably just print a DriverStation warning.

TheTripleV, 2023-12-14:
Teams might create pid controllers in commands so throwing would hurt. Clamping to zero is better imo

calcmogul, 2023-12-14:
If we do clamp to zero, we should print a DS warning so it's not a silent breaking change.

virtuald, 2023-12-14:
> Teams might create pid controllers in commands so throwing would hurt. Clamping to zero is better imo

I think you should throw if the FMS isn't connected, otherwise clamp and warn.

calcmogul, 2023-12-14:
I like that idea, but we'll have to add IsFMSConnected() to MathShared.h, since wpimath doesn't directly depend on WPILib or the HAL.

m10653, 2023-12-14:
Note that if you make Pid controllers throw an error for negitive gains you would likely also want to change the behavior of the spendable builders to clamp to zero/ignore the output.  This was done for the Izone value in https://github.com/wpilibsuite/allwpilib/pull/5959 as it would cause continuous crashes if an invalid value was set in networktables.

PR #6055 (2023-12-23): [wpimath] Report error on negative PID gains
Fixes #6045.
closed 2023-12-23 by PeterJohnson