wpilibsuite/allwpilib #6046: Make feedforward classes throw exceptions for negative Kv or Ka

calcmogul, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
Negative Kv and Ka feedforward gains create invalid, unstable models.

**Describe the solution you'd like**
Feedforward class constructors should throw for negative Kv or Ka gains.

ncorrea210, 2023-12-22:
Currently, kV and kA in the feedforward classes are public. If a check were to be implemented, it would mean either making these variables private which means adding some set functions for each, or it means checking the values every time calculate() is called. I would think making them private is better, however I'm not sure how that might interact with the protobuffs(I don't know much about them).

KangarooKoala, 2023-12-22:
For Java, `kV` and `kA` in [ArmFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/controller/ArmFeedforward.java), [ElevatorFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/controller/ElevatorFeedforward.java), and [SimpleMotorFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/controller/SimpleMotorFeedforward.java) are `public final`, so they can't be changed after construction, and [DifferentialDriveFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/controller/DifferentialDriveFeedforward.java), [LinearPlantInversion](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/controller/LinearPlantInversionFeedforward.java), and [ControlAffinePlantInversion](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/controller/ControlAffinePlantInversionFeedforward.java) have no public members (and they don't use `kV` and `kA` anyways).

However, you are right that in C++, `kV` and `kA` in [ArmFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/native/include/frc/controller/ArmFeedforward.h), [ElevatorFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/native/include/frc/controller/ElevatorFeedforward.h), and [SimpleMotorFeedforward](https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/native/include/frc/controller/SimpleMotorFeedforward.h) are public and mutable. Since Java doesn't support changing the values after construction, making them private and adding getters (and replacing the variable accesses with calls to the getters) would probably work.

PR #6084 (2023-12-23): [wpimath] Make feedforward classes throw exceptions for negative Kv or Ka
Resolves #6046.
closed 2023-12-23 by PeterJohnson