wpilibsuite/allwpilib #6047: [wpilibcExamples] Using inline specifier for Constants.h files

ncorrea210, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
When declaring global variables, while constexpr suggests to the compiler that the value may be evaluated at compile time such that a symbol is never placed on the stack, it does not require this. In instances where this does not happen, a separate version of the same value may be created for every translation unit where the "Constants.h" file is included. 

**Describe the solution you'd like**
Adding the inline specifier ensures that in ever translation unit where the "Constants.h" (or any other constants file) is included will not create a new symbol on the stack for every time it is included. Only one will be created and accessed across all units in which the constant is included. This simply requires the addition of the "inline" keyword in front of every value in Constants.h. It will not change any functionality, just save some memory(potentially). 

I am not 100% sure if something like this is worth implementing and wanted to see what others thought. If this is something worth doing, let me know and I can create a pull request doing this.

PeterJohnson, 2023-12-14:
This is absolutely the correct thing to do.  Thanks for noticing it!

virtuald, 2023-12-14:
C++ is awful

PR #6049 (2023-12-15): [wpilibcExamples] Add "inline" specifier to constants.h constants
Resolves #6047 

Note:
No adjustments made to sim examples because they used static specifier. This will ensure there is a new symbol of the same value push on the stack for every unit where the Constants.h file is included, however, I do not want to mess with those files because I don't know enough about sim. Let me know if those should be switched to 'inline constexpr' from 'static constexpr'
closed 2023-12-15 by PeterJohnson