wpilibsuite/allwpilib #6058: Easy place to find classes with structs and values

katzuv, 2023-12-17:
It'll be nice to have a list of classes that support structs, and what values they hold. It looks like the only way to do this currently is to look through the source code.

That will help with a few things. First, just to let people know this feature exists. Second, for example, we needed to dig through the source code to find that `Rotation2d` has a struct and to find out it holds the value in radians and not in degrees. It'd be easier if a list was available in, say, the API docs.

_Related: https://github.com/wpilibsuite/frc-docs/issues/2444_

PeterJohnson, 2023-12-23:
While the marker interface will help with discovery of what types support serialization, documenting the individual struct members and units thereof is not yet done.

KangarooKoala, 2023-12-24:
> documenting the individual struct members and units thereof is not yet done

So there is expected to be people writing their own struct implementations?

PeterJohnson, 2023-12-24:
Sure, teams are free to write their own.  That’s the power of it; dashboards will be able to understand even team implementations.

katzuv, 2023-12-24:
> Sure, teams are free to write their own.  That’s the power of it; dashboards will be able to understand even team implementations.

Is it possible to change the structs of classes which already have one? For example, we'd prefer `Rotation2d` to save values in degrees instead of radians.

PeterJohnson, 2023-12-24:
It’s possible to add a custom serialization for any type in Java, but not yet in C++ (until #6086 is merged). Doing so won’t affect higher level structs (eg Pose2d will still use the radian version of Rotation2d), so you’d need to do the customization all the way up the chain if you’re using those higher level types as well.  In the custom serialization, you should make sure to use a different type string as well (eg “Rotation2dDeg”), so the two versions won’t be conflated on NT or DataLog.

The basic concept for doing this in Java is to create a Struct implementation (ala Rotation2dStruct) and use that instead of Rotation2d.struct.

katzuv, 2023-12-24:
@jwbonner, provided using AdvantageKit we don't need to pass the struct object manually, is it possible to override WPILib's struct?