wpilibsuite/allwpilib #6061: wpimath execution failed

alphan3231, 2023-12-19:
* What went wrong:
Execution failed for task ':wpimath:generateProto'.
> java.io.IOException: Cannot run program "/home/elliot/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protoc/3.21.12/62e794ebd383cad86347240f1c9b419c8d8fce10/protoc-3.21.12-linux-x86_64.exe": error=2, No such file or directory

I'm using nixOS, when I want to use './gradlew build' i took that error, what should i do?

calcmogul, 2023-12-19:
Does the file exist there? We recently stopped generating files during the build, so idk why it requires protoc by default anymore.

PeterJohnson, 2023-12-19:
We still generate the C++ protobuf files during the build.  If the file exists there, this sounds like a potential libc incompatibility?

alphan3231, 2023-12-20:
yes the file exist in '/home/elliot/.gradle/caches/modules-2/files-2.1/com.google.protobuf/protoc/3.21.12/62e794ebd383cad86347240f1c9b419c8d8fce10/'

sciencewhiz, 2023-12-20:
Can you execute that file?

auscompgeek, 2023-12-20:
Since this is NixOS, I'm guessing some library dependency can't be found. You might need to build within a FHS user env.

Could you run `lddtree` on that binary? That'll show what libraries are missing.

closed 2024-03-08 by calcmogul