wpilibsuite/allwpilib #6064: SimHooks.stepTiming SIGSEGV

truher, 2023-12-19:
**Describe the bug**
Using SimHooks.stepTiming() in tests produces segv's often, but not always.

I'm not an expert at reading hs_err files, but it seems like maybe the problem is in Notifier.cpp in WaitNotifiers()?

For my tests, I don't actually care about notifier behavior, all I want is to change what getFPGATimestamp() returns for code in the main loop.  Is there a simpler way to do that?

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Project Information:
Project Version: 2024.1.1-beta-3
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.25.1
Java Debug Extension Version: 0.55.0
Java Dependencies Extension Version 0.23.3
Java Version: 17
Java Location: /home/joel/wpilib/2024/jdk
Vendor Libraries:
    ChoreoLib (2024.0.4)
    NavX (2024.0.1-beta-4)
    CTRE-Phoenix (v5) (5.32.0-beta-1)
    CTRE-Phoenix (v6) (24.0.0-beta-2)
    REVLib (2024.0.0)
    WPILib-New-Commands (1.0.0)

**Additional context**

Here's the hs_err file, truncated to fit this form:

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f0053497ef4, pid=411573, tid=411579
#
# JRE version: OpenJDK Runtime Environment Temurin-17.0.8.1+1 (17.0.8.1+1) (build 17.0.8.1+1)
# Java VM: OpenJDK 64-Bit Server VM Temurin-17.0.8.1+1 (17.0.8.1+1, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  [libc.so.6+0x97ef4]  pthread_mutex_lock+0x4
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport -p%p -s%s -c%c -d%d -P%P -u%u -g%g -- %E" (or dumping to /home/joel/FRC/TRUHER/all24/lib/core.411573)
#
# If you would like to submit a bug report, please visit:
#   https://github.com/adoptium/adoptium-support/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -Djava.library.path=/home/joel/FRC/TRUHER/all24/lib/build/jni/release -Djunit.jupiter.extensions.autodetection.enabled=true -Dorg.gradle.internal.worker.tmpdir=/home/joel/FRC/TRUHER/all24/lib/build/tmp/test/work -Dorg.gradle.native=false -javaagent:/home/joel/FRC/TRUHER/all24/lib/build/tmp/.cache/expanded/zip_0bd0851504e6286cb5fc6f37a81e25cc/jacocoagent.jar=destfile=build/jacoco/test.exec,append=true,inclnolocationclasses=false,dumponexit=true,output=file,jmx=false -Xmx512m -Dfile.encoding=UTF-8 -Duser.country=US -Duser.language=en -Duser.variant -ea worker.org.gradle.process.internal.worker.GradleWorkerMain 'Gradle Test Executor 2'

Host: Intel(R) Core(TM) i5-9400F CPU @ 2.90GHz, 6 cores, 62G, Ubuntu 22.04.3 LTS
Time: Tue Dec 19 09:26:54 2023 PST elapsed time: 1.164053 seconds (0d 0h 0m 1s)

---------------  T H R E A D  ---------------

Current thread (0x00007f004c0294c0):  JavaThread "Test worker" [_thread_in_native, id=411579, stack(0x00007f0051f00000,0x00007f0052000000)]

Stack: [0x00007f0051f00000,0x00007f0052000000],  sp=0x00007f0051ffae18,  free space=1003k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libc.so.6+0x97ef4]  pthread_mutex_lock+0x4

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  edu.wpi.first.hal.simulation.SimulatorJNI.stepTiming(J)V+0
j  edu.wpi.first.wpilibj.simulation.SimHooks.stepTiming(D)V+12
j  org.team100.lib.commands.drivetrain.DriveInALittleSquareTest.testSteering()V+165
v  ~StubRoutines::call_stub
j  jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Ljava/lang/reflect/Method;Ljava/lang/Object;[Ljava/lang/Object;)Ljava/lang/Object;+0 java.base@17.0.8.1
[... 157 lines cut ...]
0x00007f0053497f24:   10 81 e6 80 00 00 00 75 0c 64 8b 04 25 18 00 00
0x00007f0053497f34:   00 85 c0 74 37 31 c0 ba 01 00 00 00 f0 0f b1 17
0x00007f0053497f44:   0f 85 ae 00 00 00 8b 4f 08 85 c9 0f 85 5c 01 00
0x00007f0053497f54:   00 64 8b 04 25 d0 02 00 00 89 47 08 83 47 0c 01
0x00007f0053497f64:   90 31 c0 48 83 c4 10 5b c3 0f 1f 00 44 8b 07 45
0x00007f0053497f74:   85 c0 75 c1 c7 07 01 00 00 00 eb ca 8b 47 10 f6
0x00007f0053497f84:   c4 03 74 08 f6 c4 01 74 95 eb 07 90 80 cc 01 89
0x00007f0053497f94:   47 10 8b 57 10 48 83 c4 10 48 8d 77 16 5b 81 e2
0x00007f0053497fa4:   80 00 00 00 e9 63 8d ff ff 0f 1f 00 e9 eb f7 ff
0x00007f0053497fb4:   ff 0f 1f 00 81 fa 00 01 00 00 74 d6 8b 57 10 83
0x00007f0053497fc4:   e2 7f 83 fa 01 0f 85 95 00 00 00 64 8b 04 25 d0
0x00007f0053497fd4:   02 00 00 39 47 08 75 30 8b 47 04 83 f8 ff 0f 84
0x00007f0053497fe4:   bf 00 00 00 83 c0 01 89 47 04 31 c0 e9 72 ff ff 

Stack slot to memory mapping:
stack at sp + 0 slots: 0x00007f0024fb7be1: _ZN3hal13WaitNotifiersEv+0x0000000000000071 in /home/joel/FRC/TRUHER/all24/lib/build/jni/release/libwpiHal.so at 0x00007f0024f80000
stack at sp + 1 slots: 0x00007f0051ffae60 is pointing into the stack for thread: 0x00007f004c0294c0
stack at sp + 2 slots: 0x00007f0051ffaeb0 is pointing into the stack for thread: 0x00007f004c0294c0
stack at sp + 3 slots: 0x00007f0025056fc0: <offset 0x00000000000d6fc0> in /home/joel/FRC/TRUHER/all24/lib/build/jni/release/libwpiHal.so at 0x00007f0024f80000
stack at sp + 4 slots: 0x00007f0051ffae01 is pointing into the stack for thread: 0x00007f004c0294c0
stack at sp + 5 slots: 0x00007f0051ffaf80 is pointing into the stack for thread: 0x00007f004c0294c0
stack at sp + 6 slots: 0x00007f00250ca730: Java_edu_wpi_first_hal_simulation_SimulatorJNI_stepTiming+0x0000000000000000 in /home/joel/FRC/TRUHER/all24/lib/build/jni/release/libwpiHaljni.so at 0x00007f0025086000
stack at sp + 7 slots: 0x00007f0051ffae60 is pointing into the stack for thread: 0x00007f004c0294c0

[hs_err_pid411573.log](https://github.com/wpilibsuite/allwpilib/files/13718790/hs_err_pid411573.log)

calcmogul, 2023-12-19:
Maybe it's trying to lock a mutex that no longer exists? Sim notifiers have internal race conditions we were never able to figure out that cause CI to fail sometimes.

truher, 2024-01-02:
I upgraded to beta 4 and now this happens 100% of the time I use SimHooks.stepTiming or SimHooks.stepTimingAsync.

... which is not great for my test plan.

actually it also segv's when i call getNumNotifiers, whether i use the vscode test runner or the WPILib one.

is there a way to twiddle the fpga time in tests without touching this notifier problem?

calcmogul, 2024-01-03:
The SimHooks functions are the only way I know of.

calcmogul, 2024-01-03:
Could you post a test project so we can try reproducing it?

truher, 2024-01-03:
sure thing, it's here:

https://github.com/Team100/all24/tree/main/studies/simhooks_segv

it is exactly the RobotBase skeleton template, with the addition of this line in Main.main()

    NotifierSim.getNumNotifiers();

Then click "simulate robot code" and you get the segv:

# C  [libc.so.6+0x97ef4]  pthread_mutex_lock+0x4

Adding the same line to a vacuous unit test also produces the same outcome.

this is Ubuntu 22.04.3, WPI 2024 beta 4.

thanks for looking into it!

calcmogul, 2024-01-03:
I was able to reproduce it in that project with `./gradlew simulateJava` as well.

calcmogul, 2024-01-03:
Don't you have to initialize the HAL before calling HAL things?

calcmogul, 2024-01-03:
Works fine if I move that call into the Robot constructor, which is after HAL.initialize() is called.

truher, 2024-01-03:
ah, yeah, same, that fixes it.

i guess i didn't make the connection that SimHooks.stepTiming was a HAL thing at all, since i don't care about the notifier part of it, i'm really just trying to change the FPGA timestamp so that the Timer class reads the right thing.

so maybe this is WAI and there could be some sort of note about calling HAL.initialize?

truher, 2024-01-03:
oh also the original issue was a unit test issue, using the vscode test runner.  i've had issues with HAL initialization using that method in the past, which is why i knew about HAL.initialize() at all.  in the other cases, the error message was something more helpful, like "could not initialize hal" or "can't find hal jni" or whatever.  it didn't let the jvm crash.

closed 2024-02-01 by truher