wpilibsuite/allwpilib #6075: Improved WPILibC command based syntax

spacey-sooty, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
Commands in WPILibC have ugly syntax

**Describe the solution you'd like**
WPILibC command based syntax could be improved using C++ operator overloading similar to what 4788 does in behaviours. 

**Additional context**
For examples see [our docs](https://4788-docs.vercel.app/guides/behaviours/)

Starlight220, 2023-12-21:
Can you explain more specifically what operators you're suggesting with what meanings? And what is clunky/ugly in the current syntax?

Also, currently command syntax in Java and C++ look very similar and is very readable.

spacey-sooty, 2023-12-21:
New syntax wise I you can see rough examples in the docs I linked. I find the huge usage of lambda functions and requirement of calling a ton of functions to schedule them to be very clunky and ugly compared to what we have in Behaviours in 4788's codebase and conversation on discord led me to believe it might be a good idea to at least provide the option to use a somewhat similar syntax in WPILibC commands

Starlight220, 2023-12-21:
I'm not sure what you're referring to by

> requirement of calling a ton of functions to schedule them

What you have with `Behavior`s seems very much like command classes, which is still very much a viable option. And there are quite some lambdas in the docs you've linked.

In general, other than operator overloading (which is a whole other debate), the `Behavior` framework you've linked looks very similar to command-based code from ~2021, and is still very much possible today.

ncorrea210, 2023-12-22:
Though I think it may look nicer to overload operators, and perhaps it can be done ontop of the way things are currently done, I think having a more verbose syntax such as what is currently used may be better considering the potentially low entry level this library is meant to target.

spacey-sooty, 2023-12-22:
> Though I think it may look nicer to overload operators, and perhaps it can be done ontop of the way things are currently done, I think having a more verbose syntax such as what is currently used may be better considering the potentially low entry level this library is meant to target.

Yeah the idea of the operator overloading was just that it would provide a more concise option for those who wanted it but wouldn't be mandatory

closed 2024-01-17 by Starlight220