wpilibsuite/allwpilib #6076: AdvantageKit not available in StartTool Menu

3538code, 2023-12-21:
**Describe the bug**
WPIlib docs say that AdvantageKit should be in the `Start Tool` menu but I am not seeing it listed. 

**To Reproduce**
Follow instructions on the WLIB docs: https://docs.wpilib.org/en/latest/docs/software/dashboards/advantagescope.html#advantagescope

**Expected behavior**
AdvantageKit should be listed. 

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - WPILib Information:
Project Version: 2024.1.1-beta-4
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    CTRE-Phoenix (v6) (24.0.0-beta-5)
    WPILib-New-Commands (1.0.0)

**Additional context**
Perhaps the docs assume it was installed already on the system, or perhaps i need a fresh install? Not sure what the expected results should be.

- @dawonn

ncorrea210, 2023-12-22:
AdvantageKit and AdvantageScope are two seperate things, and based on what you are saying it sounds like you are refering to AdvantageScope. It appears that AdvantageScope has not yet been implemented into StartTools so for now, you can just download AdvantageScope [here](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v3.0.0-beta-5). Hope this helps!

PeterJohnson, 2023-12-22:
We install it with the WPILib installer, but haven’t yet written the pieces needed for it to show up in the tools menu.  A shortcut is installed in the WPILib Tools folder in the desktop however.

sciencewhiz, 2024-01-01:
duplicate of https://github.com/wpilibsuite/vscode-wpilib/issues/631

closed 2024-01-01 by sciencewhiz