wpilibsuite/allwpilib #6080: Better error message/handling for composing duplicate commands

rzblue, 2023-12-22:
Currently, composing the same command multiple times using constructors/factories will cause an IllegalArgumentException in `CommandScheduler.registerComposedCommands` due to set semantics. It's not immediately clear what caused it as the exception occurs multiple levels below the user's code.

We also should test this behavior - composing the same command twice should fail.

KangarooKoala, 2023-12-22:
Additionally, `ParallelDeadlineGroup` should error if a command is both the deadline and registered separately.

Starlight220, 2023-12-23:
> ParallelDeadlineGroup should error if a command is both the deadline and registered separately

The deadline is marked as composed separately from the other commands, so it does error.

PR #6091 (2023-12-27): [cmd] Improve error message when composing commands twice in same composition
Fixes #6080

Also makes `ParallelDeadlineGroup` error when the deadline appears in the other commands (with corresponding docs changes) and fixes an error message.

I looked into C++, but it doesn't have the exact same issue since registering commands as composed is done on the particular command, not on a map in the scheduler. I'm also not sure how the ownership semantics affect this issue.
closed 2023-12-27 by PeterJohnson