wpilibsuite/allwpilib #6096: Colors are not exact RGB values

scarmain, 2023-12-26:
**Describe the bug**
The RGB values that are rounded are not rounding properly.  For example, Color.kRed returns red = 1, green = 2.44140625E-4, blue = 2.44140625E-4.

I think this has to do with the Color.roundAndClamp(), this should probably only be run with the equals() function, not all the colors.

**To Reproduce**
(Using java test)
assertEquals(1, Color.kRed.red, 1e-4);
assertEquals(0, Color.kRed.green, 1e-4);  //fails, 2.44140625E-4
assertEquals(0, Color.kRed.blue, 1e-4);  //fails, 2.44140625E-4

**Expected behavior**
The values should match 1,0,0 exact

**Desktop (please complete the following information):**
 - OS: Win 10
 - Project Information:
Project Version: 2024.1.1-beta-3
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-3
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    PathplannerLib (2024.0.0-beta-6)
    CTRE-Phoenix (v5) (5.32.0-beta-2)
    CTRE-Phoenix (v6) (24.0.0-beta-4)
    REVLib (2024.0.0)
    WPILib-New-Commands (1.0.0)

**Additional context**
I was trying to make a Color.getHSV function, and it was throwing off results (red saturation would be 254.9377, calculated max(rgb)-min(rgb)/max * 255)

calcmogul, 2023-12-26:
From https://github.wpilib.org/allwpilib/docs/development/java/edu/wpi/first/wpilibj/util/Color.html:
> Represents colors.
>
> Limited to 12 bits of precision.

scarmain, 2023-12-26:
Also related, the fromHSV has a ton of rounding also, which doesn't need it, as Color accepts doubles...  (although you would have to divide by 256 for all of them)

Color.fromHSV(115, 23, 15).toHexString() = #0E0F0F
From Website https://www.rapidtables.com/convert/color/hsv-to-rgb.html  (to use this, the hue must be doubled from 180* to 360*, and the sat/value should x*100/255, as we use 255 as max)
hsv(230,9.02,5.88) = #0E0E0F

calcmogul, 2023-12-26:
We'd accept a PR that fixes the HSV conversion accuracy.

PR #6100 (2023-12-26): [wpilib] Fix precision issue in Color round-and-clamp
Fixes #6096.
closed 2023-12-26 by PeterJohnson