wpilibsuite/allwpilib #6104: Add HSV components to Color + improve color accuracy

scarmain, 2023-12-27:
Branch of #6096, this has a couple fixes.  I'm putting out this feeler PR to make sure it went the right direction before the formal request.  (and I know I have to do a C++ version too)

First, I got the color HSV accuracy to 10-bit accuracy.  The roundAndClamp function seems to add 2 bits of error to the math that I can't get out.

Second, it adds HSV components to the color components.  If you see the test cases, the outputs match exactly in doubles (no rounding error at all).

This change looks slightly weird as Java requires if you do constructor cascading, it MUST be the first instruction in the function.  It means that like the hexString constructor, I cheated using the Color8Bit implementation and came back to Color constructor to get around it  (otherwise you need some weird helper functions).  I also rearranged the constructors to have them all generate the HSV at construction.

Proposed change (and I know I have to do a C++ version too):  https://github.com/wpilibsuite/allwpilib/compare/main...scarmain:allwpilib:colorhsv

scarmain, 2023-12-27:
Also, the test cases look weird, as you are dividing by 255, not 256 here like every other program, so 128 is not color 0.5, it's actually 0.50196...

PeterJohnson, 2023-12-27:
There are many color spaces.  Wouldn’t it be cleaner for Color to remain just RGB and add a HSVColor, with conversion functions between the color classes as appropriate?