wpilibsuite/allwpilib #6109: Sending a StructTopic and a SendableChooser over NT causes sim crash on future launches

superpenguin612, 2023-12-29:
**Describe the bug**
If a StructTopic (any struct) and a SendableChooser are sent over NT4 and the SendableChooser is displayed as a window in simgui, future simulation launches will immediately crash when simgui opens.

**To Reproduce**
Run the following code in `robotInit`:
```java
StructTopic<Pose2d> topic = NetworkTableInstance.getDefault().getTable("test").getStructTopic("testPose",
        Pose2d.struct);
StructPublisher<Pose2d> publisher = topic.publish();
publisher.set(new Pose2d());

SendableChooser<String> chooser = new SendableChooser<String>();
chooser.setDefaultOption("test", "test");
chooser.addOption("test2", "test2");
SmartDashboard.putData("chooser", chooser);
```

**Expected behavior**
simgui does not crash and displays the chooser correctly

**Desktop (please complete the following information):**
 - OS: macOS 14.2.1
 - Project Information: 
```
WPILib Information:
Project Version: 2024.1.1-beta-4
VS Code Version: 1.85.1
WPILib Extension Version: 2024.1.1-beta-4
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: /Users/davidracovan/wpilib/2024/jdk
Vendor Libraries:
    WPILib-New-Commands (1.0.0)
```

**Additional context**
Error:
```
********** Robot program startup complete **********
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x000000018c038d48, pid=39293, tid=259
#
# JRE version: OpenJDK Runtime Environment GraalVM CE 22.3.1 (17.0.6+10) (build 17.0.6+10-jvmci-22.3-b13)
# Java VM: OpenJDK 64-Bit Server VM GraalVM CE 22.3.1 (17.0.6+10-jvmci-22.3-b13, mixed mode, sharing, tiered, jvmci, jvmci compiler, compressed oops, compressed class ptrs, g1 gc, bsd-aarch64)
# Problematic frame:
# C  [libc++.1.dylib+0x1ad48]  std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&)+0x14
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/XXX/Downloads/choosertest/hs_err_pid39293.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/oracle/graal/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
zsh: abort      /usr/bin/env       frc.robot.Main
```
[log file](https://github.com/wpilibsuite/allwpilib/files/13796494/hs_err_pid39293.log)

sciencewhiz, 2024-01-03:
Not related to the bug, but it looks like the JRE isn't the WPILib provided version. How are you launching simulation?

PeterJohnson, 2024-01-03:
I believe this was fixed by #6059.

closed 2024-01-03 by PeterJohnson