wpilibsuite/allwpilib #6130: Camera started with `CameraServer.startAutomaticCapture(camera)` isn't listed under CameraPublisher in NetworkTables

chauser, 2024-01-01:
**Describe the bug**
If I start a camera server with 
```
UsbCamera camera = new UsbCamera("First Camera", 0); 
MjpegServer server = CameraServer.startAutomaticCapture(camera);
```
there is no camera listed under the `CameraPublisher` key in NetworkTables -- indeed there is no `CameraPublisher` key at all.

**To Reproduce**
Start a camera server as part of a robot program with the code above. Connect to port 1181 and observe that the camera is working. Observe in Glass or OutlineViewer that there is no `CameraPublisher` key or camera-specific key for the camera.

**Expected behavior**
There should be an entry for the camera below the `CameraPublisher` key.

**Environment:**
```
WPILib Information:
Project Version: 2024.1.1-beta-4
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1-beta-4
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: /home/hauser/wpilib/2024/jdk
Vendor Libraries:
    NavX (2024.0.1-beta-3)
    PathplannerLib (2024.0.0-beta-4)
    CTRE-Phoenix (v5) (5.32.0-beta-1)
    CTRE-Phoenix (v6) (24.0.0-beta-2)
    REVLib (2024.0.0)
    WPILib-New-Commands (1.0.0)
```
-  Ubuntu 22.04
-  Testing with simulation on the desktop (but I have no reason to think the problem is related to using simulation)

**Additional context**
My goal with starting the camera this way was to be able to configure both the camera `VideoMode` settings and the stream default settings. The only obvious way to get your hands on the `MjpegServer` object where the stream default parameters can be configured is by using the `startAutomaticCapture` overload that has a VideoSource parameter and returns an `MjpegServer`. 

While writing this issue report I figured out another possible way to gain access to the `MjpegServer` object:
```
UsbCamera camera = new UsbCamera("First Camera", 0); // the string is an arbitary name, 0 is the device number
MjpegServer server = CameraServer.addServer("First Camera");
server.setSource(camera);
```
but this approach as well leads to no `CameraPublisher` NT entry although there is a server running on port 1181 and if I change the camera VideoMode setting and the stream default settings with calls on the `UsbCamera` and `MjpegServer` objects the displayed video behaves as expected.

The problem posed by this is that there seems to be no way to set the default stream parameters -- which is potentially a convenient way to ensure consistent video stream behavior across different dashboard instances.

PeterJohnson, 2024-01-01:
Duplicate of #1549; the real fix that is needed is #5135.

chauser, 2024-01-01:
I think the real fix is actually #5055, not 5135?
I am closing this issue. If anyone else runs into this, a workaround is:
```
MjpegServer server = CameraServer.addServer("First Camera");
UsbCamera cam = new UsbCamera("First Camera", 0); 
server.setSource(cam);
```
That is, explicitly create the server before creating the camera.

closed 2024-01-01 by chauser