wpilibsuite/allwpilib #6156: SmartDashboard reports wrong instance ID

Starlight220, 2024-01-05:
Resource and instance IDs are set as such:
https://github.com/wpilibsuite/allwpilib/blob/19cb2a8eb42b5a1db6c81cdafaee79a7af8ed6b6/hal/src/generate/ResourceType.txt#L44
https://github.com/wpilibsuite/allwpilib/blob/19cb2a8eb42b5a1db6c81cdafaee79a7af8ed6b6/hal/src/generate/Instances.txt#L47

Yet both C++ and Java report SmartDashboard with instance id 0:
https://github.com/wpilibsuite/allwpilib/blob/19cb2a8eb42b5a1db6c81cdafaee79a7af8ed6b6/wpilibc/src/main/native/cpp/smartdashboard/SmartDashboard.cpp#L86
https://github.com/wpilibsuite/allwpilib/blob/19cb2a8eb42b5a1db6c81cdafaee79a7af8ed6b6/wpilibj/src/main/java/edu/wpi/first/wpilibj/smartdashboard/SmartDashboard.java#L69

Gold856, 2024-01-05:
It's reported 0 for a long time. 2023 usage reporting registers `1` for all Java teams and most C++ teams for both Shuffleboard and SmartDashboard. I don't even know how that makes sense, but that's how that works.

Starlight220, 2024-01-05:
I think it registers `1` because it's a count of the reported indexes (and not the indexes themselves). Or maybe the constant here doesn't match whatever aggregates the data.

(That's why I fear that #6158 won't report accurately in cases teams trigger the LW reporting but not the SmartDashboard reporting)

PR #6157 (2024-01-05): [wpilib] Correct SmartDashboard usage reporting
Fixes #6156
closed 2024-01-05 by PeterJohnson