wpilibsuite/allwpilib #6172: Unresolved dependency: org.junit.platform junit-platform-launcher

hongyuehao, 2024-01-08:
**Describe the bug**
When I open a new project or import a project from an older version, it has this error. 

**The error code**

```
[{
	"resource": "*/Documents/FRC_Local/11/build.gradle",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": "0",
	"severity": 8,
	"message": "Unresolved dependency: org.junit.platform junit-platform-launcher",
	"source": "Java",
	"startLineNumber": 1,
	"startColumn": 1,
	"endLineNumber": 1,
	"endColumn": 1
}]
```

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 14.2.1
 - Project Information: no other extensions or venders, only a new project.

hongyuehao, 2024-01-08:
According to my friends, it is the same on windows.

ThadHouse, 2024-01-08:
See https://www.chiefdelphi.com/t/wpilib-blog-2024-kickoff-release-of-wpilib/448056/9?u=thad_house for the fix. We'll have this up in Known Issues soon. And the fix will be in the next wpilib release as well.

handsomepwy, 2024-01-08:
> According to my friends, it is the same on windows.

yeah it's me on win10:)

closed 2024-01-08 by hongyuehao