wpilibsuite/allwpilib #6175: `SysIdRoutine` creates log files when `recordState` is overridden

jwbonner, 2024-01-08:
Deferred from #6033, see comment [here](https://github.com/wpilibsuite/allwpilib/pull/6033#discussion_r1442459590). `DataLogManager` is started even if the `recordState` handle is overridden, which can create empty log files unnecessarily (even if the routines are instantiated but never run). One example where this is problematic is for AdvantageKit users, who may want to log the state to an AdvantageKit log instead of using `DataLogManager`. I've temporarily fixed the problem in that case using AdvantageKit's shims, but this is not ideal.

The best solution is probably to defer creation of the state log entry until `recordState` is called (see example [here](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/junction/shims/wpilib/src/edu/wpi/first/wpilibj/sysid/SysIdRoutineLog.java#L228) from the shimmed version).

calcmogul, 2024-01-08:
Keep in mind that you'll need AdvantageKit/Scope to export a wpilog for SysId to ingest.

jwbonner, 2024-01-08:
AdvantageKit already records to wpilog, it just handles the data flow independently. The example projects show how to record the state through AdvantageKit, and the inputs are already logged in the subsystems (or they could be exported through a vendor log).

PR #6259 (2024-01-20): Defer creation of state log entry in `SysIdRoutineLog`
Fixes #6175
closed 2024-01-20 by PeterJohnson