wpilibsuite/allwpilib #6176: Provide ARM64 Linux builds

CoffeeCoder1, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
To install WPILib on an aarch64 Linux machine, such as a Raspberry Pi or an Apple Silicon computer, the user has to build WPILib themselves.

**Describe the solution you'd like**
Provide aarch64 binaries of the WPILib installer.

TheTripleV, 2024-01-08:
A raspberry pi image is provided by wpilib.

An apple silicon version of the installer is available alongside the downloads for the other platforms.

CoffeeCoder1, 2024-01-08:
The Raspberry Pi image is only supported on Raspberry Pi computers, which does not fix the problem of installing on an Apple Silicon or similar computer running Linux (in my case, I have a MacBook on which I run the Asahi Linux kernel).

ThadHouse, 2024-01-12:
The changes needed to the installer are a bit too invasive to do this season, but the PR has been pulled up to date and should be available for 2025.

https://github.com/wpilibsuite/WPILibInstaller-Avalonia/pull/305

ThadHouse, 2024-01-22:
@CoffeeCoder1 https://github.com/wpilibsuite/allwpilib/releases/tag/v2024.2.1 now has arm64 builds. We don't have a way to test them, so can't guarantee everything about it works, but most things should.

closed 2024-01-22 by PeterJohnson