wpilibsuite/allwpilib #6177: [Glass/simgui] 2024 field image is not positioned correctly for coordinate system

shueja, 2024-01-08:
**Describe the bug**
[image]
Also, the field dimensions show as 2023's dimensions when 2024 is selected 
**To Reproduce**
- Open a field visualization in Glass or SimGui (this probably requires making a robot project with a Field2d published)
- Set the image to 2024.

sciencewhiz, 2024-01-09:
Reopening because with #6179 it looks like the robot still isn't placed correctly

[image]

jasondaming, 2024-01-10:
Is there a way to change this locally until this change gets pushed?

sciencewhiz, 2024-01-10:
There's two issues. There's problems with the field border in the 2024 json file. #6185 fixes that. There also appears to be a bug in the glass scaling code that is yet unfound. So while you could use the files in #6185 locally for pathweaver, it will still have problems with glass.

sciencewhiz, 2024-01-14:
The combination of #6185 and #6222 fix this.

PR #6179 (2024-01-09): Fix 2024 field json
field corners and field size were identical to 2023
Fixes #6177
closed 2024-01-14 by sciencewhiz