wpilibsuite/allwpilib #6178: Documentation installed by WPILibInstaller points to beta Javadocs

fovea1959, 2024-01-08:
**Describe the bug**
After installing 2024.1.1, the local copy of "2024 WPILib Documentation" has a link for WPILib Java API docs that points to the beta Javadocs at "https://github.wpilib.org/allwpilib/docs/beta/java/index.html".

**To Reproduce**
Steps to reproduce the behavior:
1. Install 2024.1.1.
2. Open the "2024 WPILib Documentation". Navigate to "WPILib Java API docs", select it. 

**Expected behavior**
Expect to link to local copy of 2024.1.1 javadocs at "file:///C:/Users/Public/wpilib/2023/documentation/java/index.html" instead of cloud copy of 2024.1.1-beta-4 javadocs at "https://github.wpilib.org/allwpilib/docs/beta/java/index.html".

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Project Information: N/A.

**Additional context**
None.

sciencewhiz, 2024-01-09:
Can you expand on what you mean by: 
> Open the "2024 WPILib Documentation". Navigate to "WPILib Java API docs", select it.

On my 2024.1.1 installation, in VS Code I click on the WPI icon, choose, `WPILib: Open API Documentation` then choose `java`, it opens the local documentation. I'm assuming you're going a different way, but not sure exactly what it is.

sciencewhiz, 2024-01-09:
I figured it out, it's the desktop shortcut of 2024 WPILib Documentation, which opens the local readthedocs copy.

ThadHouse, 2024-01-09:
That means this will automatically be fixed when we do the next release. Its because the installer was built before readthedocs was updated.

sciencewhiz, 2024-01-09:
Readthedocs should have been updated, the PR to update was merged hours before the installer was built. But it should be fine for the next release because readthedocs os updated now.