wpilibsuite/allwpilib #6188: [sysid] Failure to analyze AdvantageKit log files

jwbonner, 2024-01-09:
The attached log file was produced using AdvantageKit in simulation, with very minor noise added to the voltage samples to ensure reasonable sampling for the dynamic tests (even with duplicate values removed). Trying to run the SysId analysis using the fields below produces the message "Not enough data to run Acceleration Calculation..." on Windows and causes the application to crash on macOS (not tested on Linux).

- Test state: `/RealOutputs/SysIdState`
- Velocity: `/Drive/LeftVelocityRadPerSec`
- Position: `/Drive/LeftPositionRad`
- Voltage: `/Drive/LeftAppliedVolts`
- Config: simple, radians, velocity/position scaling of 1

The console log for the macOS crash shows:

```
INFO: Preparing data (Analyzer.cpp:263)
INFO: Preprocessing raw data. (AnalysisManager.cpp:119)
INFO: Copying raw data. (AnalysisManager.cpp:121)
zsh: segmentation fault  /Users/jonah/wpilib/2024/tools/SysId.app/Contents/MacOS/sysid
```

[SysId_AdvantageKit.wpilog.zip](https://github.com/wpilibsuite/allwpilib/files/13878778/SysId_AdvantageKit.wpilog.zip)

Running with WPILib 2024.1.1 on Windows 10 and macOS 13.5.

PeterJohnson, 2024-01-10:
Other than the datalog loading functionality, it appears the SysIdState in this file doesn't use the correct naming.  What the tool expects is lowercase-names (e.g. quasistatic-forward, dynamic-reverse), but what's in the file is kQuasistaticForward, kDynamicReverse.

PeterJohnson, 2024-01-10:
@Oblarg any thoughts here?

Oblarg, 2024-01-10:
For now we should reject anything that doesn't have the standard naming.  We can become more flexible later when we improve the fit.

PeterJohnson, 2024-01-10:
Ok. Currently there's just the 4 valid names? (quasistatic and dynamic / forward and reverse combinations)

Oblarg, 2024-01-10:
And `none`, yeah.

PR #6200 (2024-01-12): [sysid] Filter valid test names
Currently the analysis portion only supports quasistatic and dynamic, forward and reverse. Check for anything not matching and remove it, along with providing diagnostics of what is being loaded.

Fixes #6188.
closed 2024-01-12 by PeterJohnson