wpilibsuite/allwpilib #6194: 2024 field image json - appears to still be incorrect

jsimpso81, 2024-01-10:
**Describe the bug**
After the update the JSON file for the 2024 field image appears to still be incorrect.

The value 1008 probbly should be 1108.  

Also curious how the values were derived?   

In looking at a zoomed in file, it seems that values should be 112, 27, 1108, 522.   Except for the 1008, these are only very slightly different than the JSON.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the field in one of the applications that displays fields and visually review.

 - Link to code:
https://github.com/wpilibsuite/allwpilib/blob/main/fieldImages/src/main/native/resources/edu/wpi/first/fields/2024-crescendo.json

**Expected behavior**
The entire field should be viewable.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - Windows 10  (Used Paint to view zoomed in image..)
 - N/A
 - 
**Additional context**
Add any other context about the problem here.

sciencewhiz, 2024-01-11:
Yes the 1008 is a typo, #6185 changes to a better image and fixes the dimensions. I suspect that the +/- 2 pixel is just a difference in exactly where it's measured, since there's a gradient.

shueja, 2024-01-12:
> Also curious how the values were derived?

Typically, one uses the ancient method of looking at the pixel coordinates of the cursor in MS Paint to find the coordinates of each wall.

closed 2024-01-14 by PeterJohnson