wpilibsuite/allwpilib #6204: Glass.app does not exist

a1cd, 2024-01-12:
**Describe the bug**
When running Gradle `:Glass` on MacOS, the program fails, saying 
```
Executing 'Glass'...

> Task :Glass

BUILD SUCCESSFUL in 413ms
1 actionable task: 1 executed
The file ~/wpilib/2024/tools/Glass.app does not exist.
Execution finished 'Glass'.
```

Glass.app does not exist in the `~/wpilib/2024/tools/` folder. The `glass` executable is there, along with `Glass.py` and `Glass.vbs`.

**To Reproduce**
Steps to reproduce the behavior:
1. Install WPILib on intel or ARM macbook (select (fastest) in the installer)
2. run the Gradle task `Glass` (or `simulateJava` which doesn't make anything appear)

 - Link to code:

**Expected behavior**
No errors when opening glass, and when simulating, the glass simulation thing that normally opens doesn't open (note, I am not using vscode, I use intelij. For the glass simulation popup, using vscode instead of gradle would make it work properly I believe).

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: MacOS
 - Project Information: [In Visual Studio Code, press the WPILib button and choose WPILib: Open Project Information. Press the copy button and paste the data here. If not using VS Code, please include WPILib version, Gradle version, Java version, C++ version (if applicable), and any third party libraries and versions]
```
$ ./gradlew -v

------------------------------------------------------------
Gradle 8.4
------------------------------------------------------------

Build time:   2023-10-04 20:52:13 UTC
Revision:     e9251e572c9bd1d01e503a0dfdf43aedaeecdc3f

Kotlin:       1.9.10
Groovy:       3.0.17
Ant:          Apache Ant(TM) version 1.10.13 compiled on January 4 2023
JVM:          20.0.2 (Oracle Corporation 20.0.2+9-78)
OS:           Mac OS X 14.1.1 x86_64
```
I use intelij, and in intelij (which runs gradle on whatever java is in intelij), I configured using the official `temurin-11` JVM from the WPILib 2024 folder. Language level is 11. Anything run in intelij is run using these settings.

ThadHouse, 2024-01-12:
Glass is separate from sim gui, so the simulate java issue is different.

Where did you get the project from? All the projects created from VS Code have `wpi.sim.addGui().defaultEnabled = true`. The `defaultEnabled = true` part of that is what makes `simulateJava` show the gui. Without that, the sim gui will not be loaded.

I'll look into the glass issue. But its separate.

trevnels, 2024-01-12:
This project is based on one of the 2024 AdvantageKit templates. Its `build.gradle` has `wpi.sim.addGui()`, but is missing the `.defaultEnabled = true`. It sounds like that addition should fix things, but I'll let @a1cd confirm.

jwbonner, 2024-01-12:
AdvantageKit intentionally disables the simulation GUI by default, since it can't be used when running a log replay. For example, this allows you to run a log replay on the command line without using VSCode to explicitly disable the GUI. Of course, you can always add `defaultEnabled = true` manually if you prefer to open the GUI for physics simulation.

We'll add a comment in `build.gradle` for the next release that explains the reasoning behind the change.

PR #6286 (2024-01-22): Fix macOS apps not always being an application
Closes #6204

#5586 was grabbing the incorrect executable on macOS. Also for some reason I only applied that to Glass and RTNS, when it should have been applied to all tools. Fix Glass and RTNS, and update the other tools to fix the issue that PR was originally supposed to fix.
closed 2024-01-22 by PeterJohnson