wpilibsuite/allwpilib #6205: Datalog Tool 120fps option doesn't set to 120fps.

Daltz333, 2024-01-12:
- 30fps is really 20fps
- 120fps is ~65fps

https://github.com/wpilibsuite/allwpilib/assets/10674555/d3c0a215-812f-493c-a8c6-893ec26ba02b

PeterJohnson, 2024-01-12:
All of the imgui based tools use the same code for FPS limiting: https://github.com/wpilibsuite/allwpilib/blob/main/wpigui/src/main/native/cpp/wpigui.cpp#L366-L373

I'm not sure how we can make this code any more robust?

Gold856, 2024-09-18:
We might want to take some inspiration from MjpegServer and use average frame times + `continue` with faster loops to get more precise frame times. 
https://github.com/wpilibsuite/allwpilib/blob/main/cscore/src/main/native/cpp/MjpegServerImpl.cpp#L727