wpilibsuite/allwpilib #6215: DriverStation.isAStopped(); Request

jpace3937, 2024-01-13:
I was looking for a method to get a boolean that would inform me if our robot is autonomous stopped, similar to DriverStation.isEStopped();.

Is there any way that this method could be added?

Thanks so much for all the work you do!

ThadHouse, 2024-01-13:
That data isn't actually sent from the DS to the Robot. All the robot sees is it has been disabled. It looks identical to the period of time the robot is disabled for between autonomous and teleop.

TheTripleV, 2024-01-13:
You could probably use a mix of reading the match time and isdisabled as a heuristic for a stopped.

ThadHouse, 2024-01-13:
I'm pretty sure GetMatchTime is going to return 0 when disabled, just like the period between auto and teleop. All a stop looks like is a longer gap between auto and teleop.

closed 2024-01-13 by PeterJohnson