wpilibsuite/allwpilib #6228: Add Simulation Init to Subsystems

m10653, 2024-01-14:
**Is your feature request related to a problem? Please describe.**
Currently there only exists a method to periodically update simulation objects. This requires the creation of simulation objects either regardless of if the robot is in simulation,  or extra logic to check if it is in simulation before creating these objects.

**Describe the solution you'd like**
It would be nice if there was an additional overridable method added to Subsystems that only runs once at robot start when the robot is indeed being simulating, allowing for simulation specific object to be Initialized. 

If this is a reasonable feature to add I have no problem creating a PR to add this feature.

Starlight220, 2024-01-15:
Partial duplicate of #4148

closed 2024-01-15 by Starlight220

Starlight220, 2024-01-15:
See my comment on #4148:

> In general, the problem to do this is that `Subsystem` is currently used for multiple different things, where it's intended to be primarily for hardware mutexing.
> Adding more methods increases the significance of other purposes, which isn't necessarily something we want to do.
> 
> 
> Additionally, adding these methods to the subsystem in team code is quite easy, and even more so in the discussed (2025?) inlining of RobotContainer into Robot.
> As for `simInit`, an `if (RobotBase.isSimulation)` block at the end of the constructor accomplishes the task without much verbosity and with more flexibility for hardware-attached simulation.