wpilibsuite/allwpilib #6229: Kotlin Gradle DSL roborioDebug and roborioRelease not found

supercoolspy, 2024-01-15:
When using the kotlin gradle dsl (build.gradle.kts) roborioDebug and roborioRelease are not found. They do not seem to be available, I tried doing project("roborioDebug") and it was unable to find it

**To Reproduce**
Steps to reproduce the behavior:
1. Download bad-build.gradle.kts
2. Add it to the timed robot example

 - Link to code: https://gist.github.com/supercoolspy/c37b104ee5d9774a02d0a6a1aa5f6f70

**Desktop (please complete the following information):**
 - OS: nixos
 - Project Information: 
   - WPLib Version: 2024.1.1
   - Java Version: Zulu 17
   - Gradle: 8.5

ThadHouse, 2024-01-15:
We do not support the kotlin DSL. I would highly recommend not using it, we have no plans to support it in the future at all.

https://github.com/wpilibsuite/GradleRIO/blob/c7a425a39dc479f3528a3737ec65d7d5445c2d4f/src/main/java/edu/wpi/first/gradlerio/deploy/roborio/FRCJavaArtifact.java#L42

Both of those are just configuration names, I'm not sure if there's anything in the Kotlin DSL needed to access those.

supercoolspy, 2024-01-15:
> We do not support the kotlin DSL. I would highly recommend not using it, we have no plans to support it in the future at all.

Is there a reason or is it just not planned?

I'll see if I can fix it thank you for the info

supercoolspy, 2024-01-15:
> It's not planned, because every time I try to use it I want to rip my eyes out. We actually used to have some projects using it, and moved back to groovy because it was unmaintainable.

Fair, do you have any guides on groovy that are good used to using kotlin dsl

ThadHouse, 2024-01-15:
Not really. Just leaned it through years of dealing with it. I know gradle's docs always show examples in both.

tylerlowrey, 2024-01-24:
@supercoolspy You should be able to do:

```kotlin
  wpi.java.deps.wpilibJniDebug(NativePlatforms.roborio).forEach {
      add("roborioDebug", it)
  }
  wpi.java.vendor.jniDebug(NativePlatforms.roborio).forEach {
      add("roborioDebug", it)
  }

  wpi.java.deps.wpilibJniRelease(NativePlatforms.roborio).forEach {
      add("roborioRelease", it)
  }
  wpi.java.vendor.jniRelease(NativePlatforms.roborio).forEach {
      add("roborioRelease", it)
  }
```

I've got a working (with java simulation at least, haven't tested on a RoboRIO yet) build.gradle.kts here: https://github.com/tylerlowrey/kotlin-robot-code-2024/blob/main/build.gradle.kts

closed 2024-03-08 by calcmogul