wpilibsuite/allwpilib #6236: new python versions of Ctre and Wpilib cannot be installed at the same time

Lagor845, 2024-01-16:
**Describe the bug**
Every time I install the python ctre module it installs the old version of wpilib.

**To Reproduce**
Steps to reproduce the behavior:
You basically install ctre and then install wpilib. I have had this happen on multiple computers. 

Any tips to get us limping in the meantime?
Thank you for your help!

virtuald, 2024-01-16:
Please post your python version, the output of pip list, and what commands you've been running.

Be sure to use the 2024 instructions at https://docs.wpilib.org/en/stable/docs/zero-to-robot/step-2/python-setup.html

sciencewhiz, 2024-01-20:
Note that the name changed this year. robotpy-ctre will install last year's version.

virtuald, 2024-01-20:
There's now a `phoenix6` package, but `robotpy-ctre` still exists (but it provides `phoenix5` and not `ctre` like it used to).

closed 2024-03-08 by calcmogul