wpilibsuite/allwpilib #6250: Gradlew returns an error when building but gradle builds perfectly fine

cherriae, 2024-01-18:
**Describe the bug**
A clear and concise description of what the bug is.

The bug is that when I create a new project and it automatically runs `gradlew build  -Dorg.gradle.java.home="C:\Users\Public\wpilib\2024\jdk" ` but instead on building it doesn't build and returns a error around `org.gradle.launcher.GradleMain`

**To Reproduce**
Steps to reproduce the behavior:
1. Download wpilib 2024.1.1 (windows 11)
2. Create a new project
3. Then build code

**Expected behavior**
A clear and concise description of what you expected to happen.

I expected that it would build without errors but it builded with these errors: 

```
Executing task: gradlew build   -Dorg.gradle.java.home="C:\Users\Public\wpilib\2024\jdk" 

Exception in thread "main" java.lang.ClassNotFoundException: org.gradle.launcher.GradleMain
        at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:445)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:592)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:525)
        at org.gradle.wrapper.GradleWrapperMain.main(SourceFile:70)

 *  The terminal process "cmd.exe /d /c gradlew build   -Dorg.gradle.java.home="C:\Users\Public\wpilib\2024\jdk"" terminated with exit code: 1. 
```

I've tried to give `gradlew` commands like `--help` or `--debug` but all it does is build for some reason.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

This is building with `gradle` (not gradlew):
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Project Information:
```
WPILib Information:
Project Version: 2024.1.1
VS Code Version: 1.85.1
WPILib Extension Version: 2024.1.1
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    WPILib-New-Commands (1.0.0)
```
**Additional context**
Add any other context about the problem here.

I installed `gradle` on my own computer but it builds perfectly but `./gradlew` doesn't build. I've redownload wpilib mulitple times and I've tried my arch linux and github codespaces which both were able to build except my windows.

ThadHouse, 2024-01-18:
Do you have any antivirus software on the computer, and is it a school managed computer?

If the answer to both of those is no, can you delete the .gradle folder in your home directory? Before you do, reboot your computer otherwise it won’t be able to delete it.

cherriae, 2024-01-18:
> Do you have any antivirus software on the computer, and is it a school managed computer?
> 
> If the answer to both of those is no, can you delete the .gradle folder in your home directory? Before you do, reboot your computer otherwise it won’t be able to delete it.

This is a personal laptop and I have deleted `.gradle` folder multiple times but nothings working I've tried reboot my computer many times. I'm not sure but If this is because of microsoft defender antivirus.

statulr, 2024-01-18:
@cherriae can't believe I found another arch user that is in FRC

ThadHouse, 2024-01-18:
> > Do you have any antivirus software on the computer, and is it a school managed computer?
> > If the answer to both of those is no, can you delete the .gradle folder in your home directory? Before you do, reboot your computer otherwise it won’t be able to delete it.
> 
> This is a personal laptop and I have deleted `.gradle` folder multiple times but nothings working I've tried reboot my computer many times. I'm not sure but If this is because of microsoft defender antivirus.
> 
> I have even tried regenerating the gralde wrapper with gradle but nothing seems to working
> 
> Directory structure: [tree.txt](https://github.com/wpilibsuite/allwpilib/files/13971802/tree.txt)

You're deleting the .gradle folder in your home folder, right? Not the .gradle folder in the project.

cherriae, 2024-01-19:
@ThadHouse Thank you so much, it seems to be an issue with my cache in my home directly prbly something downloaded wrong

closed 2024-01-19 by cherriae