wpilibsuite/allwpilib #6254: Riolog (in VSCode) displays nothing until and unless the DriverStation is connected

chauser, 2024-01-19:
**Describe the bug**
The Riolog window (in VSCode) displays nothing until and unless the DriverStation is connected.

**To Reproduce**
Close the DriverStation if one is already open
Deploy Robot code
Observe Riolog window opens but nothing appears in the Riolog
Start the DriverStation and make sure it connects
Observe output is now flowing to the Riolog
Close the DriverStation
Observe that output to Riolog stops
 - Link to code:
I don't think this is code dependent
**Expected behavior**
Riolog gets robot program output regardless of connection state of the DriverStation.
The behavior seems to be the same regardless of whether the team number is configured in the Riolog pane.
I tried it with the 2023 VSCode/Riolog -- the behavior is the same, but I don't remember it acting this way last year.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 11] - Windows 11 (haven't checked on Windows 10)
 - Project Information: WPILib Information:
Project Version: 2024.1.1
VS Code Version: 1.85.1
WPILib Extension Version: 2024.1.1
C++ Extension Version: 1.19.1
Java Extension Version: 1.26.2023121408
Java Debug Extension Version: 0.55.2023121302
Java Dependencies Extension Version 0.23.2023120100
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    NavX (2024.1.0)
    PathplannerLib (2024.1.3)
    CTRE-Phoenix (v6) (24.1.0)
    photonlib (v2024.2.0)
    REVLib (2024.2.0)
    WPILib-New-Commands (1.0.0)

**Additional context**
This is frustrating because you miss all the output from the beginning of the execution of the robot program. I am also unable to Restart Robot Code from the driverstation (as reported earlier for the beta) so I really don't know how to capture the output from the beginning of the robot program.
I don't know if these issues are related -- but if they are I expect they point to the driverstation code rather than the riolog code.

ThadHouse, 2024-01-19:
This behavior has been there for years. In Netcomm, the TCP listeners are not created until the DS is connected. So we can't solve this from our end, it would be something NI would need to fix, and it would be a massive overhaul of Netcomm to do so.

chauser, 2024-01-19:
Okay. Tells you how good my memory is from one year to the next.
Any thoughts on the inability to restart robot code?

ThadHouse, 2024-01-19:
Which issue did you create for the restart robot code issue? I don't see one.

chauser, 2024-01-19:
Well, I see now it wasn't an issue but just a discussion in the beta project: https://github.com/wpilibsuite/2024Beta/discussions/54 and I hadn't realized that it always happens. I can make a new issue for it now and then this one can be closed.

chauser, 2024-01-19:
Closing this. The "can't restart robot code" problem is not reproducible at this point.

chauser, 2024-01-19:
Behaving as expected. Won't fix.

closed 2024-01-19 by Starlight220