wpilibsuite/allwpilib #6257: Generic sanity test on hardware ports

NiranSchneller, 2024-01-19:
**Describe the bug**
When deploying code that tries to initialize a PWM Port that is <= 0, the roboRIO doesn't do anything upon ENABLE, and when trying to redeploy code, it says build failed because of - discoverroborio.

**To Reproduce**
Steps to reproduce the behavior:
1. Initialize a PWM Port which is <=0
2. Deploy code
3. Try to redeploy code, and see that the build fails on - discoverroborio.

 - Link to code:

**Expected behavior**
The expected behavior is getting an error that the port we entered was invalid.

**Desktop (please complete the following information):**
 - OS: Windows 11.
 - Project Information: 

Project Version: 2024.1.1
VS Code Version: 1.84.0
WPILib Extension Version: 2024.1.1
C++ Extension Version: 1.17.5
Java Extension Version: 1.23.0
Java Debug Extension Version: 0.52.0
Java Dependencies Extension Version 0.23.0
Java Version: 17
Java Location: C:\Users\Public\wpilib\2024\jdk
Vendor Libraries:
    NavX (2024.1.0)
    CTRE-Phoenix (v5) (5.33.0)
    CTRE-Phoenix (v6) (24.1.0)
    REVLib (2024.2.0)
    WPILib-New-Commands (1.0.0)

**Additional context**

It started when we had problems deploying our robot code. The code deployment failed on discoverroborio, so e naturally though it was a networking issue of some sort. After endless pings, network checks and version switches (2023 and back), we realized they weren't the problem, and the problem had to be in the code itself. After many more hours of debugging, we found out the problem was that we had initialized a PWM Port of 0 in the code. While that fixed it, I still could not find find any correlation between the problem (Invalid port) and the error that we got, since we should have gotten an error about the port. So out of curiosity, I looked at the JNI files in this repository. After close inspection, I found out that there is no checking for PWM ports that are <= 0. Not checking this is what caused the RIO to crash everytime we tried to redeploy code, when the previous deployed code had that incorrect port.

ThadHouse, 2024-01-19:
The port validation actually happens here

https://github.com/wpilibsuite/allwpilib/blob/a2d45dbca406ee7bdc17d1da010c509006606dd2/hal/src/main/native/athena/HAL.cpp#L120

NiranSchneller, 2024-01-20:
Ok,
You are correct, I apologize for the misunderstanding. After further debugging, I found out that the problem indeed wasn't in the port. The problem was this:

We initialized an AddressableLEDBuffer of length zero called - ledBuffer.
Then we ran the following line:
this.led.setData(this.ledBuffer);

this.led is of type AddressableLED.

This means that from what I understand, it is still a problem on the WPI side.
If I am wrong, please correct me and I will do further debugging to address the issue.

ThadHouse, 2024-01-20:
What error are you seeing in that case? That’s a much different case than originally reported. Can you post screenshots?

NiranSchneller, 2024-01-20:
What happens is that the code deploys very slowly, and then when trying to redeploy discoverroborio fails.
In terms of debugging, how I originally fixed the problem was commenting the whole LED class, which in terms of fixing the issue worked, but not in terms of discovering the root cause. So what I did was go into the subsystem, and comment the constructor. Then I started commenting the lines, and the line I uncommented that caused the problem was:

this.led.setData(this.ledBuffer);
When the ledBuffer is of length 0, like I said.

Regarding errors, the error from code deployment was:

Output:
admin @ 172.22.11.2: Resolved but not connected.
Reason: RuntimeException
com.jcraft.jsch.JSchException: timeout: socket is not established
admin @ null: Resolved but not connected.
Reason: TimeoutException
Discovery timed out.
admin @ roborio-3339-FRC.local: Resolved but not connected.
Reason: TimeoutException
Discovery timed out.
admin @ roborio-3339-FRC: Resolved but not connected.
Reason: TimeoutException
Discovery timed out.
admin @ 10.33.39.2: Resolved but not connected.
Reason: TimeoutException
Discovery timed out.
admin @ roborio-3339-FRC.lan: Failed resolution.
Reason: RuntimeException
Unknown Host
admin @ roborio-3339-FRC.frc-field.local: Failed resolution.
Reason: RuntimeException
Unknown Host
Run with --info for more details

Task :discoverroborio FAILED
Missing Target!
=============================================
Are you connected to the robot, and is it on?
=============================================
GradleRIO detected this build failed due to not being able to find “roborio”!
Scroll up in this error log for more information.

FAILURE: Build failed with an exception.

What went wrong:
Execution failed for task ‘:discoverroborio’.
A failure occurred while executing edu.wpi.first.deployutils.deploy.target.discovery.TargetDiscoveryWorker
Target roborio could not be found at any location! See above for more details.

Try:
Run with --stacktrace option to get the stack trace.
Run with --info or --debug option to get more log output.
Run with --scan to get full insights.
Get more help at https://help.gradle.org/.

BUILD FAILED in 4s
4 actionable tasks: 2 executed, 2 up-to-date

Keep in mind that to be able to redeploy code, I had to manually remove the .jar file by manually SSHing into the RIO, and restarting the bot to reset the RAM and be able to redeploy.

I have looked at the code and drawn the following conclusion:

https://github.com/wpilibsuite/allwpilib/blob/main/hal/src/main/native/cpp/jni/AddressableLEDJNI.cpp#L75

What happens as a result from this line is that we will point to memory address 0 because of the length of jArrBuf, thus crashing the RIO.

sciencewhiz, 2024-01-20:
I took the AddressableLED Java example and changed it to `m_ledBuffer = new AddressableLEDBuffer(0);` and commented out the call to the rainbow method and duplicated the issue.

ThadHouse, 2024-01-20:
Weird. There must be a bug in the FPGA that causes a 0 length string to spin the CPU (If you look at CPU utilization in the DS it's 100% and pegged). Easy to work around on our end, just don't pass data to the FPGA on a 0 length string.

PR #6271 (2024-01-20): Don't write a 0 length led string to the FPGA
Due to something weird in the FPGA, calling strobeLoad() with a string length of 0 causes both CPUs to spin at 100%, basically shutting down everything else on the robot.

If a 0 length string happens to be passed, just bail out early.

Closes #6257
closed 2024-01-20 by PeterJohnson