wpilibsuite/allwpilib #6260: Driver Station Timing Window always says good timing

sciencewhiz, 2024-01-20:
I used [clumsy](https://jagt.github.io/clumsy/index.html) to simulate a bad network connection. I set it to throttle 20% of packets for 100ms. The DS lost communication about every second. However, the Timing Good LED on the timing window always stayed green.

[image]

chauser, 2024-01-20:
And the x-axis has no time scale making it hard to cross-reference information displayed here to anything else.

jashow55, 2024-01-21:
I am trying to debug possible DS disables due to tighter tolerances in 2024 DS:  #6174 

I just wanted to say I also see no x-axis time scale.
This is first time using DS Log File Viewer for 2024. I am using Win11.  When I resize the window by grabbing a corner with mouse and left click, I briefly see the x-axis scale but it seems to zoom too much to fit the new window size.  I have yet to try another laptop.

sciencewhiz, 2024-01-21:
The log viewer scaling issue is #6181

sciencewhiz, 2024-02-05:
Timing is still always marked as good in patch 1

Gold856, 2026-05-05:
OBE by the new DS.

closed 2026-05-05 by Gold856