wpilibsuite/allwpilib #6275: Add shim for C++23's std::expected

calcmogul, 2024-01-21:
It would be nice to have a shim for C++23's std::expected, since our compilers won't all support it for [a while](https://en.cppreference.com/w/cpp/compiler_support/23). It's a convenient alternative to exceptions and is an analog for Rust's Result type.

We could make an upstream_utils script grab https://github.com/TartanLlama/expected/blob/master/include/tl/expected.hpp and replace `namespace tl` with `namespace wpi`.

ncorrea210, 2024-01-23:
This does seem like a nice thing to have. Just wondering if there is any concern about using differing error handling methods between C++ and Java? I imagine java will remain using exceptions.

calcmogul, 2024-01-23:
Divergence there is fine, imo. We try to follow each language's idioms (i.e., have the same featureset, but different implementation details). We use exceptions in Java because it's the only realistic option; Optional has object allocation overhead. C++ has cleaner error handling alternatives at this point (outside of constructors; Rust got it right by returning a Result from the constructor).

ncorrea210, 2024-01-23:
Fair enough. I do like Rust's "Result" as a way of handling errors. I was just unsure if a change such as this may go against WPILib's goal of ensuring both C++ and Java are supported equally.

PR #6310 (2024-01-26): [upstream_utils] Add std::expected shim
Its tests use Catch2 instead of GoogleTest, so we can't import them.

Fixes #6275.
closed 2024-01-26 by PeterJohnson