wpilibsuite/allwpilib #6288: Error building networktables with cmake

JakeStew81, 2024-01-22:
**Describe the bug**
When attempting to install only networktables using cmake on a coprocessor, I ran into the following error when trying to build:

    pathToRepo/allwpilib/wpinet/src/main/native/cpp/UDPClient.cpp: In member function 'int wpi::UDPClient::set_timeout(double)`:
    pathToRepo/allwpilib/wpinet/src/main/native/cpp/UDPClient.cpp:240:18: error: aggregate 'wpi::UDPClient::set_timeout(double)::timeval tv' has incomplete type and cannot be defined
      240 |    struct timeval tv;
          |                   ^˜

**To Reproduce**
Steps to reproduce the behavior:
1. Clone WPI Lib repository
2. Make a build directory
3. Run: `cmake ../allwpilib/ -DWITH_JAVA=OFF -DWITH_JAVA_SOURCE=OFF -DWITH_CSCORE=OFF -DWITH_WPIMATH=OFF -DWITH_WPIUNITS=OFF -DWITH_EXAMPLES=OFF -DWITH_TESTS=OFF -DWITH_GUI=OFF -DWITH_SIMULATION_MODULES=OFF -DWITH_WPILIB=OFF`
4. Run `make -j4`
5. Comment out most of Stacktrace.cpp because alpine doesn't have execinfo
6. Re-run `make -j4`
7. Get the error shown above

 - Link to code: N/A

**Expected behavior**
`make` to build networktables

**Desktop (please complete the following information):**
 - OS: Alpine Linux

**Additional context**
On a coprocessor, a [Beelink Mini S12](https://www.bee-link.com/beelink-mini-s12-n95-mini-pc-clone-1) to be exact

ThadHouse, 2024-01-22:
What version of gcc is that using? And also we don’t do any musl testing, so it’s possible this is a difference between musl and glibc.

JakeStew81, 2024-01-22:
gcc version 13.2.1, I just installed it with apk

PeterJohnson, 2024-01-22:
Add an `#include <sys/time.h>` to that file, and see how much further you get.  There will probably be other issues with musl vs glibc, as we don't test compilation with musl.  I just opened #6288 to fix this particular case, but I'm guessing there are others.

JakeStew81, 2024-01-22:
Didn't end up with any other errors, although I was just doing networktables, nothing else. Thanks for the help!

PR #6289 (2024-01-22): [build] Fix compilation with musl
Fixes #6288.
closed 2024-01-22 by JakeStew81