wpilibsuite/allwpilib #6302: Glass Constant Disconnects: ill-formed UTF-8 byte/nonzero RSV

jonathandao0, 2024-01-24:
**Describe the bug**
Glass clients keep disconnecting repeatedly when attempting to connect to the robot. This happens the instant a Glass client connects to the robot. This appears to only occur with the real robot and not if the code is simulated. Attempting to switch between NT4/NT3 does not appear to fix the issue This occurs across multiple computers. It appears that the issue does not affect Windows 10 computers and the computers affected so far appear to be only Windows 11 computers. Using other NT programs (SmartDashboard, AdvantageScope) is not affected. This affects both WPILib 2024.1.1 and 2024.2.1 releases of Glass.

Glass NetworkTables Debug Log:

> starting network client (NetworkClient.cpp:45)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 77221: syntax error while parsing object key - invalid string: ill-formed UTF-8 byte; last read: '"typ�'; expected string literal (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid literal; last read: '"msgpack"}},�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid literal; last read: '"msgpack"}},�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 80081: syntax error while parsing array - invalid literal; last read: '"msgpack"}}�'; expected ']' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 80081: syntax error while parsing array - invalid literal; last read: '"msgpack"}}�'; expected ']' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 77221: syntax error while parsing object key - invalid string: ill-formed UTF-8 byte; last read: '"�'; expected string literal (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 80081: syntax error while parsing object key - invalid string: ill-formed UTF-8 byte; last read: '"param�'; expected string literal (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing object key - invalid string: ill-formed UTF-8 byte; last read: '"type�'; expected string literal (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing object key - invalid string: ill-formed UTF-8 byte; last read: '"type�'; expected string literal (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
[... 41 lines cut ...]
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
[... cut ...]
WARNING: could not decode JSON message: [json.exception.parse_error.101] parse error at line 1, column 74361: syntax error while parsing value - invalid string: ill-formed UTF-8 byte; last read: '"i�' (WireDecoder.cpp:116)
DISCONNECTED NT4 connection: nonzero RSV (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
DISCONNECTED NT4 connection: remote end closed connection (NetworkClient.cpp:458)
CONNECTED NT4 to 10.42.1.2 port 5810 (NetworkClient.cpp:402)
DISCONNECTED NT4 connection: connection timed out (NetworkClient.cpp:458)

PeterJohnson, 2024-01-24:
What version of robot code?

jonathandao0, 2024-01-24:
Is this good enough?

[glassWiresharkCapture.zip](https://github.com/wpilibsuite/allwpilib/files/14032267/glassWiresharkCapture.zip)

sciencewhiz, 2024-01-24:
Assuming this is your code, it's still using 2024.1.1

https://github.com/4201VitruvianBots/Crescendo2024

jonathandao0, 2024-01-24:
Yeah, we realized that the WPILib version in build.gradle wasn't updated yet. After updating it, we haven't seen the issue.

closed 2024-01-24 by jonathandao0