wpilibsuite/allwpilib #6308: Unit Safe PID

spacey-sooty, 2024-01-25:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I mess up units in my code.

**Describe the solution you'd like**
Using the C++ units library it is possible to create a unit safe PID controller using generics.

**Additional context**
This is definitely plausible as we have done it in our teams library Wombat

calcmogul, 2024-01-25:
We originally didn't make PIDController use units because there is a small but non-negligible learning curve that we wanted to avoid for such a widely used class.

spacey-sooty, 2024-01-25:
If the learning curve is considered a major issue here then we could maybe add a seperate unit safe PID class? I wouldn't be opposed either way

calcmogul, 2024-01-25:
If we do have separate classes, it's not clear what to name each.

Starlight220, 2024-01-26:
Is it possible with some template magic to have the class support both units and raw doubles?

spacey-sooty, 2024-01-26:
The way I've seen it implemented in FRC libraries usually required the generic to be a unit, but hypothetically it might be possible to pass a double in and treat it as a quantity or something. I think it would depend on the internals of wpimath units though