wpilibsuite/allwpilib #6309: `TrapezoidProfileCommand` not using new `TrapezoidProfile` API correctly

R7B9, 2024-01-25:
The new `TrapezoidProfile` API introduced in the 2024 update, which gets the initial and goal state each time you call calculate instead of in the constructor seems to not be correctly used in the `TrapezoidProfileCommand`. Currently, it starts a timer at initialization, and then each cycle uses its time with the current state and goal state in the profile calculation. This is a wrong use of the API, which I think comes from a bad naming of the `calculate()` method parameter which is named "current" but should actually be "initial". This is probably because that in the old API, the initial value for the profile was actually the current one, because it was passed in the creation of the `TrapezoidProfile` object. So with the new API, because you need to pass the initial state in each calculation, you have two options:

1. Save the initial state in a variable and then use it for each `calculate()` call. If you do that, you should also reset the timer and save a new initial state each time the goal state changes (or just don't allow changing the goal state). This will maintain the same functionality of the old API.
2. Instead of passing the time passed since initialization, pass the duration of one cycle, like [shown in the documentation](https://docs.wpilib.org/en/stable/docs/software/advanced-controls/controllers/trapezoidal-profiles.html#complete-usage-example). This will change the way the old `TrapezoidProfile` worked and will essentially make it closed-loop.

I think offering a choice between these two will be the best option, as each technique is useful for different scenarios.
Either way, I think there is an improvement needed for the parameter names and Javadoc to make the the new API more clear, because as I described, it's not completely accurate and is quite confusing.

calcmogul, 2024-01-25:
It being named current is consistent with the underlying TrapezoidProfile class. The only bug here is the timer handling.

I've never been a fan of command wrappers like this because they're leaky abstractions that force the users to learn the internals anyway to use them properly.

R7B9, 2024-01-26:
I think the name `current` should be changed to `initial` in the `TrapezoidProfile` class too.

R7B9, 2024-01-26:
But isn't it the state which the trapezoidal profile is starting from? It also makes sense to me because if you want to use trapezoidal profile as open loop like before, you would need to save the initial value and pass it each time, and pass the time since the initialization. Also, maybe instead make it like with `PIDController` and the `setpoint`? So you would be able to use it without updating the profile itself.

calcmogul, 2024-01-26:
> But isn't it the state which the trapezoidal profile is starting from?

No, the initial state is the state at time t=0. The current state is where the robot _currently_ is, not where it initially was.

> if you want to use trapezoidal profile as open loop like before, you need to save the initial value and pass it each time, with the time passed since the initialization

No, you don't. See the intended usage in the [JavaDocs](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/math/trajectory/TrapezoidProfile.html).

> maybe make it like with the PIDController class and the setpoint property of it? So you would be able to call calculate() for a different time without updating the profile itself.

PIDController doesn't work that way. [ProfiledPIDController.calculate(measurement)](https://github.wpilib.org/allwpilib/docs/release/java/edu/wpi/first/math/controller/ProfiledPIDController.html#calculate(double))'s behavior seems closer to what you're asking for. TrapezoidProfile takes the current state for two reasons:

1. Functional purity reasons
2. It allows using the class as a form of feedback by passing the current state estimate instead of the current reference

R7B9, 2024-01-26:
This is a usage which is essentially closed-loop, no? So you shouldn't use trapezoidal profile as open loop anymore like it used to be? Is there a reason for it?

calcmogul, 2024-01-26:
Open-loop just means system outputs (e.g., measurements) aren't being fed back into the controller. If you provide the current setpoint as the JavaDoc example shows, that meets the definition of open-loop.

R7B9, 2024-01-26:
What's written there is what I tried to describe, sorry if I wasn't clear enough. From what's written there, it's still means that the parameter is not necessarily the current state but just the initial state for the profile being calculated. And I still think it would be cool to have the `TrapezoidProfile` class save the initial and goal states inside and just let you change it, but that's only for convenience of use.

calcmogul, 2024-01-26:
> it's still means that the parameter is not necessarily the current state but just the initial state for the profile being calculated

I just explained why that's not the case, but OK: https://github.com/wpilibsuite/allwpilib/issues/6309#issuecomment-1911253570

R7B9, 2024-01-26:
How can the caulcate method get the current state without being closed-loop? And as far as I understand it, the state you pass is the state at time=0. Sorry if I don't understand you correctly.

calcmogul, 2024-01-26:
> How can the calculate method get the current state without being closed-loop?

The current trajectory state can be either a reference (open-loop trajectory) or robot state estimate/measurement (closed-loop trajectory).

> the state you pass is the state at time=0

The initial state is the state at the initial time (t=0). The current state is the state at the current time. The initial time and current time are not the same thing. It seems like you're conflating the two usage patterns for a trapezoid profile.

R7B9, 2024-01-26:
Ok, so let's make it clear:
1. For closed-loop, you pass the current state with the time passed since the last cycle.
2. For open-loop, you pass the initial state with the time passed since the start of the movement. (initialization of it's command).

I understand why for the closed-loop, it might be easier to understand it as the current state and the time you want to advance from the current state toward the goal state. For the open-loop option, this is not very clear. Either way, I think I wasn't wrong saying that essentially, the calculation just creates a profile between the two states, and gives the state at the time you passed, this is just another way of looking at it.
BTW - Have you seen better results with the closed-loop profile than the open-loop profile (if both use both feedforward and PID for following it)?

calcmogul, 2024-01-26:
> 1. For closed-loop, you pass the current state with the time passed since the last cycle.
> 2. For open-loop, you pass the initial state with the time passed since the start of the movement. (initialization of it's command).

This not what the terminology means. Again, the only thing that determines whether it's closed-loop or open-loop is whether there's measurements being fed back. If you provide a setpoint to (1), it's open-loop.

In either case, always using the current state instead of whatever state was given when the profile was created is more robust to goal changes; there's corner cases in the profile math you can encounter otherwise.

R7B9, 2024-01-26:
> whether there's measurements being fed back

And it does if you pass the current state each cycle, the current state is a measurement in itself.

calcmogul, 2024-01-26:
> the current state is a measurement in itself

That's not what measurement means. In control systems, setpoints and measurements are different things.

R7B9, 2024-01-26:
I actually read some of it, but it gets into very complicated math very fast, making it nearly impossible for me to understand it.

R7B9, 2024-01-26:
After trying to read it again, I still don't get why isn't it measurement. You are getting the current state from the sensors. This is, according to the book definition in its glossary, is considered output/measurement.

calcmogul, 2024-01-26:
This is open-loop:
```cpp
TrapezoidProfile<units::meter>::Constraints constraints{kMaxV, kMaxA};
auto previousProfiledReference = 0_m;
TrapezoidProfile profile{constraints};

while (1) {
    previousProfiledReference = profile.Calculate(
        20_ms, previousProfiledReference, 2_m);
}
```

This is closed-loop:
```cpp
TrapezoidProfile::Constraints<units::meter> constraints{kMaxV, kMaxA};
TrapezoidProfile profile{constraints};

while (1) {
    previousProfiledReference = profile.Calculate(
        20_ms, units::meter_t{encoder.GetDistance()}, 2_m);
}
```

R7B9, 2024-01-26:
Ok, I get what you are saying now, that's kind of what I tried to say. sorry for not being clear enought. Thank you

Starlight220, 2024-01-27:
@calcmogul in your open-loop example, wouldn't the state returned from `calulate` always be the same? You're sampling the same timestep.

calcmogul, 2024-01-27:
Nope. In that example, `Calculate(t, current, goal)` returns the state after the double integrator travels for 20 ms from {0 m, 0 m/s} toward {2 m, 0 m/s} under the given dynamics constraints. The time parameter is relative to the current state, not the initial state.

`current` is set to the return value so the next call starts a little closer. This pattern essentially makes our motion profile classes filters, and they can be functionally composed like such.

Here's a worked out example. Let:
```
maxV = 1 m/s
maxA = 1 m/s²
t = 20 ms = 0.02 s
initial state = q₀ = {0 m, 0 m/s}
goal = q_f = {2 m, 0 m/s}
```

```
{x₁, v₁} = Calculate(0.02 s, {x₀, v₀}}, {x_f, v_f})
{x₁, v₁} = Calculate(0.02 s, {0 m, 0 m/s}, {2 m, 0 m/s})
{x₁, v₁} = {x₀ + v₀t + 1/2 at², v₀ + at}
{x₁, v₁} = {0.0002 m, 0.02 m/s}

{x₂, v₂} = Calculate(0.02 s, {x₁, v₁}}, {x_f, v_f})
{x₂, v₂} = Calculate(0.02 s, {0.0002 m, 0.02 m/s}, {2 m, 0 m/s})
{x₂, v₂} = {x₁ + v₁t + 1/2 at², v₁ + at}
{x₂, v₂} = {0.0008 m, 0.04 m/s}

{x₃, v₃} = Calculate(0.02 s, {x₂, v₂}}, {x_f, v_f})
{x₃, v₃} = Calculate(0.02 s, {0.0002 m, 0.02 m/s}, {2 m, 0 m/s})
{x₃, v₃} = {x₂ + v₂t + 1/2 at², v₂ + at}
{x₃, v₃} = {0.0018 m, 0.06 m/s}
```

The [docs](https://github.wpilib.org/allwpilib/docs/development/java/edu/wpi/first/math/trajectory/TrapezoidProfile.html#calculate(double,edu.wpi.first.math.trajectory.TrapezoidProfile.State,edu.wpi.first.math.trajectory.TrapezoidProfile.State)) have been updated to hopefully make this clearer.

Starlight220, 2024-01-27:
> current is set to the return value so the next call starts a little closer. 

In that case, the code in your earlier is incomplete -- is an assignment to `reference` missing?

calcmogul, 2024-01-27:
> is an assignment to reference missing?
No.
[image]

Starlight220, 2024-01-27:
Oh oops I must've missed that. In that case, it makes sense. Sorry for the confusion.

closed 2024-03-08 by calcmogul